raspiblitz/raspiblitz #2127: Sync slow process and 0 peers

tibxfool, 2021-04-02:
Goodmorning,

I am currently syncing the blockchain after a crash. When I started I had about 10 peers and now its down to 0. Process seems to be very slow now. Is this normal? Currently at 48.08% and very slowly increasing. Can I reboot the PI without having to completely start from the beginning to reconnect to the peers?  I started syncing on the 28th March and progress was lot better.

Thx.
Frits

openoms, 2021-04-02:
you can restart any time
if no peers can add the manually

tibxfool, 2021-04-02:
Thanks. Will more peers speed up the process?

openoms, 2021-04-02:
if there is 1 peer bitcoind will connect to more.
The difference is between having peers or none.
Of for some reason the peers connected are the bottleneck a restart helps.

tibxfool, 2021-04-02:
without a restart back to 11 now. :-)

Thanks for your help!

closed 2021-04-15 by rootzoll