raspiblitz/raspiblitz #2128: Issue with dphys-swapfile.service

frennkie, 2021-04-02:
@rootzoll  @openoms I just ran an apt upgrade on my test node and the process failed:

```
admin@raspiblitz~ $ sudo apt full-upgrade -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  lxplug-volume
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up dphys-swapfile (20100506-5+rpt2) ...
Job for dphys-swapfile.service failed because the control process exited with error code.
See "systemctl status dphys-swapfile.service" and "journalctl -xe" for details.
invoke-rc.d: initscript dphys-swapfile, action "restart" failed.
● dphys-swapfile.service - dphys-swapfile - set up, mount/unmount, and delete a swap file
   Loaded: loaded (/lib/systemd/system/dphys-swapfile.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-04-02 11:29:39 BST; 2s ago
     Docs: man:dphys-swapfile(8)
  Process: 14865 ExecStart=/sbin/dphys-swapfile setup (code=exited, status=137)
 Main PID: 14865 (code=exited, status=137)

Apr 02 11:29:08 tthm3 systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Apr 02 11:29:39 tthm3 dphys-swapfile[14865]: computing size, want /mnt/hdd/swapfile=1870MByteKilled
Apr 02 11:29:39 tthm3 systemd[1]: dphys-swapfile.service: Main process exited, code=exited, status=137/n/a
Apr 02 11:29:39 tthm3 systemd[1]: dphys-swapfile.service: Failed with result 'exit-code'.
Apr 02 11:29:39 tthm3 systemd[1]: Failed to start dphys-swapfile - set up, mount/unmount, and delete a swap file.
dpkg: error processing package dphys-swapfile (--configure):
 installed dphys-swapfile package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 dphys-swapfile
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I found this post and the "solution" (`systemctl disable dphys-swapfile`) worked - although I'm not sure yet of the consequences: 
 https://github.com/nextcloud/nextcloudpi/issues/1219.

Have also seen/fixed this already?!

openoms, 2021-04-02:
Is this with the 1.7 RC1 ?
I have done apt upgrades on the 64bit base image previously had no issues.

frennkie, 2021-04-02:
No.. this is still on 1.6.3. It's fine with me to close this here.. If anybody runs into it then this either solves it or the issue can be reopened.

closed 2021-04-02 by frennkie