raspiblitz/raspiblitz #2130: bitcoind service is NOT RUNNING

Akaline66, 2021-04-02:
could someone help with what I need to do to get the service back up and running - i was about 80% done with syncing and I was moving some things around and maybe I hit a wire onthe raspiblitz node and saw the error on the screen.  

rocess: 31602 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=1/FAILURE)

Apr 02 22:47:33 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 02 22:47:33 raspberrypi systemd[1]: Failed to start Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
-- Logs begin at Thu 2021-04-01 04:49:16 BST, end at Fri 2021-04-02 22:47:43 BST. --
Apr 02 22:47:33 raspberrypi systemd[1]: Starting Bitcoin daemon...
Apr 02 22:47:33 raspberrypi bitcoind[31602]: ************************
Apr 02 22:47:33 raspberrypi bitcoind[31602]: EXCEPTION: N5boost10filesystem16filesystem_errorE
Apr 02 22:47:33 raspberrypi bitcoind[31602]: boost::filesystem::create_directory: File exists: "/home/bitcoin/.bitcoin"
Apr 02 22:47:33 raspberrypi bitcoind[31602]: bitcoin in AppInit()
Apr 02 22:47:33 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Apr 02 22:47:33 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 02 22:47:33 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
.
.
.
Apr 02 22:46:02 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Apr 02 22:46:02 raspberrypi systemd[1]: lnd.service: Failed to load environment files: No such file or directory
Apr 02 22:46:02 raspberrypi systemd[1]: lnd.service: Failed to run 'start-pre' task: No such file or directory
Apr 02 22:46:02 raspberrypi systemd[1]: lnd.service: Failed with result 'resources'.
Apr 02 22:46:02 raspberrypi systemd[1]: Failed to start LND Lightning Daemon.
Apr 02 22:47:02 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Apr 02 22:47:02 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 18.
Apr 02 22:47:02 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Apr 02 22:47:03 raspberrypi systemd[1]: lnd.service: Failed to load environment files: No such file or directory
Apr 02 22:47:03 raspberrypi systemd[1]: lnd.service: Failed to run 'start-pre' task: No such file or directory
Apr 02 22:47:03 raspberrypi systemd[1]: lnd.service: Failed with result 'resources'.
Apr 02 22:47:03 raspberrypi systemd[1]: Failed to start LND Lightning Daemon.

openoms, 2021-04-04:
Looks like you have run into a disk coruption issue:
```
]: EXCEPTION: N5boost10filesystem16filesystem_errorE
```
start again, use the recommeded parts and keep your node safe.

Akaline66, 2021-04-04:
I do have the recommended hardware (save for the screen).  I selected the option to start syncing the chain from new - and to my suprise, the machine rebooted and continued at 80%.

closed 2021-04-15 by rootzoll