raspiblitz/raspiblitz #2131: bitcoin failed to start

Bechev, 2021-04-03:
Hi there,

I'm having an issue setting up my node. I installed it a first time, copied the blockchain on the HDD, and couldn't make it run. I did a repair keeping the blockchain on the HDD and I'm still facing an error. Not sure I'm getting what the log is trying to tell me.

Does anyone have a fox for that? Thank you

Unit lnd.service could not be found.
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory
nginx: [emerg] BIO_new_file("/mnt/hdd/app-data/nginx/tls.cert") failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D080:BIO routines:BIO_new_file:no such file)
nginx: configuration file /etc/nginx/nginx.conf test failed
https://termbin.com/qxcy

maxchernin, 2021-04-03:
@Bechev  Seems like a technical issue rather than a bug
i suggest you get live help @ the telegram group of raspiblitz. if you haven't already

https://t.me/raspiblitz

openoms, 2021-04-04:
Suggest to flash the SDcard again and make sure it is a good quality one.
Use the recommended parts.

closed 2021-04-15 by rootzoll