raspiblitz/raspiblitz #2134: Make it possible to save the blockchain via samba

Psychoshybc1, 2021-04-03:
Would be good if there would be an options to save the blockchain via Samba on a network share, to make integration in an existing network easier and maybe cheaper because the requirement to buy an SSD could be obsolete.

nyxnor, 2021-04-07:
The requirement to buy a SSD would not be obsolete because
1. You still need to have storage on your drives.
2. The disk need to be formatted and partitioned.
3. Setting a dedicated machine is more secure.
4. To my knowledge samba is not encrypted.

This is why it is better to have the SSD directly attached on your node.

I dont know if by integration you mean syncing to another drive from your home or by just listening.
* If you want to [transfer the blockchain on your LAN](https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#i-have-the-full-blockchain-on-another-computer-how-do-i-copy-it-to-the-raspiblitz)
* If you want to [listen from a blockchain on your LAN.](https://github.com/openoms/bitcoin-tutorials/blob/master/joinmarket/joinmarket_desktop_to_blitz.md#lan-connection)

rootzoll, 2021-04-07:
Also keep in mind that using a network storage would be quite slow down everything. Just with the speed up from RaspberryPi3-USB2 to RaspberryPi4-USB3 plus HDD to SSD we got into the area that its possible to validate the blockchain within 2-3 days and not weeks/month as before. I would really not recommend changing back to slower baclhainstorages anymore.

closed 2021-04-07 by rootzoll