raspiblitz/raspiblitz #2138: add patch to make Tor read through folders /etc/tor/torrc.d and /etc/tor/hidden-services.d/

nyxnor, 2021-04-04:
### These are the sources dependent on:
* https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder
* https://github.com/Kixunil/debcrafter
* Im specifically talking about the package [tor-hs-patch-config](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/blob/master/pkg_specs/tor-hs-patch-config.sps) that do [these patches](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/tree/master/tor-extras/assets)

I think this helps a lot organizing Tor configuration. Best described by [Kixunil here](https://github.com/rootzoll/raspiblitz/issues/2054#issuecomment-805560571):

>Maybe you would be interested in my package called tor-hs-config-patch, which does clean patching of Tor installation to enable reading parts configuration from /etc/tot/torrc.d and /etc/tor/hidden-services.d.
The advantages of this approach:
>*    User can modify configuration by placing additional file(s) into above mentioned directories without editing the original
>*   No annoying questions about changed configuration during package upgrade
>*   Cleaner organization
>*   Trivial to remove some piece (just run rm)
>*   Various different scripts will not affect each-other, not even accidentally (unless they configure inherently conflicting options)
>*   Scripts can be made idempotent - that means running them multiple times (regardless if previous attempts failed or not) does not break anything. I do this in all my deb packages.

### My points:
* I encountered issues when updating Tor such as asking me to override or maintain previous file configuration.
* Also I found pretty useful to parse folder such as `/etc/tor/hidden-services.d/` so every single service get its separate file. It helps organization, deletion.
* I actually wanted to use this since building the sd card, as I maintained Tor on the building, so this would help a lot, as there were some workarounds if not using this patch

### Now the unfortunate part being third party:
* @Kixunil [explanation of not relying on him.](https://github.com/rootzoll/raspiblitz/issues/2054#issuecomment-811776341)
* Third party. But a way to solve this, instead of addding his repo to sources list, is you hosting the files to be visible on github, so on the build script, it is made a .deb package to be installed.
* This would be done on the build_sdcard.sh due to installing Tor since there (to reach Tor keys when it is blocked). So this will need to be scripted.
* the [config.scripts/internet.hiddenservice.sh](https://github.com/rootzoll/raspiblitz/blob/dev/home.admin/config.scripts/internet.hiddenservice.sh) would be one of the files that would change a lot, but it is a short one, should be quick to arrange it correctly to work well with the patch.

### Info
* This works on raspios_arm64 (Me and @pretyflaco tested the patch on v1.7.0RC1).
* Even if this is just for v1.8, I would want to know, cause it will modify certain parts of some important files I'm trying to pull request, such as internet.hiddenservice.sh, internet.tor.sh, build_sdcard.sh. So I dont want to pull request if this is not done, maybe if rejected, would still do this for my fork, as the benefits are clear to me.
* @Kixunil, if I made any mistakes, please correct me as always.

Kixunil, 2021-04-04:
Everything looks correct.

>  is you hosting the files to be visible on github, so on the build script, it is made a .deb package to be installed.

I guess it means you want to have what Debian calls native source package. That should work well, I believe. If accepted, I will try to produce it directly, so you don't have to depend on `debcrafter`.

The only downside of this approach is you will have to update explicitly when I release anew version but that should be negligible issue as building that package is easy and fast and I don't plan to change anything there in the near future anyway.

nyxnor, 2021-04-08:
@rootzoll ping
This could be for v1.8, no rush.

nyxnor, 2021-05-08:
Blitz does not use apparmor, so maybe this is not needed: `owner /etc/tor/** r,` for `/etc/apparmor.d/local/system_tor`

Found this to add to the first lines of the file:
```
sudo sed -i '1 i %include /etc/tor/torrc.d' /usr/share/tor/tor-service-defaults-torrc
sudo sed -i '1 i %include /etc/tor/hidden-services.d\n' /usr/share/tor/tor-service-defaults-torrc
```

Well, these are just commands, but the difference being is that does not need to ping another repo, does not have problems of package being removed by mistake.

Kixunil, 2021-05-14:
It'd also get overwritten whenever `tor` updates. It needs `dpkg-divert` to avoid such problems.

nyxnor, 2021-12-01:
as this was not analyzed to include third party package, installing tor setting --confold to avoid diversion will do for now.
`sudo apt -o Dpkg::Options::="--force-confold" install -y tor`
- #2711

rootzoll, 2022-03-09:
@nyxnor PR was merged for v1.7.2 - was there anything more or can close?

nyxnor, 2022-03-10:
related to this issue can close

closed 2022-04-05 by rootzoll