raspiblitz/raspiblitz #2144: BTC RPC Explorer - 502 Bad Gateway

pinnubi, 2021-04-05:
Hello everyone,

I have a issue with the BTC RPC Explorer, when calling the interface in my browser, I get a 502 Bad Gateway.

I restarted the service already. A bit more info:

**sudo systemctl restart btc-rpc-explorer && sudo systemctl status btc-rpc-explorer**

● btc-rpc-explorer.service - btc-rpc-explorer
   Loaded: loaded (/etc/systemd/system/btc-rpc-explorer.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-04-05 21:14:42 BST; 112ms ago
 Main PID: 23013 (btc-rpc-explore)
    Tasks: 5 (limit: 4915)
   Memory: 1.3M
   CGroup: /system.slice/btc-rpc-explorer.service
           ├─23013 /bin/bash /home/admin/system/btc-rpc-explorer.run.sh
           ├─23015 /bin/bash /home/admin/system/btc-rpc-explorer.run.sh
           ├─23016 sudo -u electrs lsof -i
           ├─23017 grep -c 50001
           └─23027 lsof -i

Apr 05 21:14:42 raspberrypi systemd[1]: Started btc-rpc-explorer.
Apr 05 21:14:42 raspberrypi btc-rpc-explorer.run.sh[23013]: Waiting Electrs on port 50001...
Apr 05 21:14:42 raspberrypi sudo[23016]:    admin : TTY=unknown ; PWD=/home/btcrpcexplorer/btc-rpc-explorer ; USER=electrs ; COMMAND=/usr/bin/lsof -i
Apr 05 21:14:42 raspberrypi sudo[23016]: pam_unix(sudo:session): session opened for user electrs by (uid=0)

**systemctl status nginx:**

● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/nginx.service.d
           └─raspiblitz.conf
   Active: active (running) since Mon 2021-04-05 00:48:22 BST; 20h ago
     Docs: man:nginx(8)
 Main PID: 974 (nginx)
    Tasks: 5 (limit: 4915)
   Memory: 10.9M
   CGroup: /system.slice/nginx.service
           ├─974 nginx: master process /usr/sbin/nginx -g daemon on; master_process on;
           ├─975 nginx: worker process
           ├─976 nginx: worker process
           ├─978 nginx: worker process
           └─979 nginx: worker process

**When running only $ sudo systemctl status btc-rpc-explorer:**

● btc-rpc-explorer.service - btc-rpc-explorer
   Loaded: loaded (/etc/systemd/system/btc-rpc-explorer.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2021-04-05 21:14:59 BST; 2min 43s ago
  Process: 23863 ExecStart=/home/admin/system/btc-rpc-explorer.run.sh (code=exited, status=1/FAILURE)
 Main PID: 23863 (code=exited, status=1/FAILURE)

**When running /home/admin/syste/btc-rpc-explorer.run.sh alone:**

sudo /home/admin/system/btc-rpc-explorer.run.sh
Waiting Electrs on port 50001...
Electrs started, launching BTC-RPC-Explorer...

> btc-rpc-explorer@2.0.0 start
> node ./bin/www

events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE: address already in use 127.0.0.1:3002
    at Server.setupListenHandle [as _listen2] (net.js:1318:16)
    at listenInCluster (net.js:1366:12)
    at doListen (net.js:1503:7)
    at processTicksAndRejections (internal/process/task_queues.js:81:21)
Emitted 'error' event on Server instance at:
    at emitErrorNT (net.js:1345:8)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  code: 'EADDRINUSE',
  errno: -98,
  syscall: 'listen',
  address: '127.0.0.1',
  port: 3002
}

Any ideas how to proceed?

Thank you very much and have a great day.

nyxnor, 2021-04-07:
Error: listen EADDRINUSE: address already in use 127.0.0.1:3002
`sudo netstat -tunlp | grep 3002`  to see what is listening on that port. Investigate by PID.
Havent seen this issue on rpc-explorer before...

Try to set `off` then `on` again.

`sudo bash config.scripts/bonus.btc-rpc-explorer.sh off`
`sudo bash config.scripts/bonus.btc-rpc-explorer.sh on`

pinnubi, 2021-04-07:
Hi, thanks for your help!

Here is some more info:

admin@raspberrypi:~ $ sudo netstat -tunlp | grep 3002
tcp        0      0 127.0.0.1:3002          0.0.0.0:*               LISTEN      1003/nginx: master
tcp6       0      0 ::1:3002                :::*                    LISTEN      1003/nginx: master

admin@raspberrypi:~ $ sudo bash config.scripts/bonus.btc-rpc-explorer.sh off
# making sure services are not running
# *** REMOVING BTC-RPC-explorer ***
Removed /etc/systemd/system/multi-user.target.wants/btc-rpc-explorer.service.
userdel: btcrpcexplorer mail spool (/var/mail/btcrpcexplorer) not found
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
# OK service is removed - restarting TOR ...
# Done
# OK BTC-RPC-explorer removed.
Rule updated
Rule updated (v6)
Rule updated
Rule updated (v6)

admin@raspberrypi:~ $ sudo bash config.scripts/bonus.btc-rpc-explorer.sh on
# making sure services are not running
# *** INSTALL BTC-RPC-EXPLORER ***
nodeJS is already installed
Installed nodeJS v14.15.4
txindex is already active
Adding user `btcrpcexplorer' ...
Adding new group `btcrpcexplorer' (1007) ...
Adding new user `btcrpcexplorer' (1007) with group `btcrpcexplorer' ...
Creating home directory `/home/btcrpcexplorer' ...
Copying files from `/etc/skel' ...
Cloning into 'btc-rpc-explorer'...
remote: Enumerating objects: 474, done.
remote: Counting objects: 100% (474/474), done.
remote: Compressing objects: 100% (298/298), done.
remote: Total 6629 (delta 304), reused 292 (delta 173), pack-reused 6155
Receiving objects: 100% (6629/6629), 10.92 MiB | 6.51 MiB/s, done.
Resolving deltas: 100% (4451/4451), done.
HEAD is now at 87ce1e4 2.0.0

added 407 packages, and audited 408 packages in 45s

9 vulnerabilities (5 low, 4 high)

To address issues that do not require attention, run:
  npm audit fix

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.
npm notice
npm notice New minor version of npm available! 7.7.6 -> 7.8.0
npm notice Changelog: https://github.com/npm/cli/releases/tag/v7.8.0
npm notice Run npm install -g npm@7.8.0 to update!
npm notice
# getting RPC credentials from the bitcoin.conf
mv: cannot move '/home/admin/btc-rpc-explorer.env' to '/home/btcrpcexplorer/.config/btc-rpc-explorer.env': No such file or directory
chown: cannot access '/home/btcrpcexplorer/.config/btc-rpc-explorer.env': No such file or directory
# *** Updating Firewall ***
Rule updated
Rule updated (v6)
Rule updated
Rule updated (v6)

nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
*** Install btc-rpc-explorer systemd ***
Created symlink /etc/systemd/system/multi-user.target.wants/btc-rpc-explorer.service → /etc/systemd/system/btc-rpc-explorer.service.
# OK - the BTC-RPC-explorer service is now enabled
# needs to finish creating txindex to be functional
# monitor with: sudo tail -n 20 -f /mnt/hdd/bitcoin/debug.log

# Hidden Service for btc-rpc-explorer
HiddenServiceDir /mnt/hdd/tor/btc-rpc-explorer
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:3022
HiddenServicePort 443 127.0.0.1:3023

Restarting Tor to activate the Hidden Service...

The Tor Hidden Service address for btc-rpc-explorer is:
xxxxxxxxxxxxxxx
use with the port: 80
or the port: 443

Apr 07 15:43:25 raspberrypi sudo[26506]:     root : TTY=unknown ; PWD=/ ; USER=btcrpcexplorer ; COMMAND=/bin/sed -i s/^BTCEXP_ADDRESS_API=none/BTCEXP_ADDRESS_API=electrumx/g /home/btcrpcexplorer/.config/btc-rpc-explorer.env
Apr 07 15:43:25 raspberrypi sudo[26506]: pam_unix(sudo:session): session opened for user btcrpcexplorer by (uid=0)
Apr 07 15:43:25 raspberrypi _background.sh[9491]: sed: can't read /home/btcrpcexplorer/.config/btc-rpc-explorer.env: No such file or directory
Apr 07 15:43:25 raspberrypi sudo[26506]: pam_unix(sudo:session): session closed for user btcrpcexplorer
Apr 07 15:43:25 raspberrypi _background.sh[9491]: electrs is active - service start via script is already enabled, nothing to do here
Apr 07 15:43:25 raspberrypi _background.sh[9491]: BTC-RPC-Explorer service needs restart

Sounds a bit like some files are missing?

nyxnor, 2021-04-07:
Two things, `npm` and `ipv6` giving problems.

[To solve `npm` you need to update it.](https://github.com/rootzoll/raspiblitz/issues/2124#issuecomment-811486504)

For solving ipv6 try this  
`sudo bash ~/config.scripts/internet.sh ipv6 on`

rootzoll, 2021-04-07:
FYI there is an update coming for BTC RPC Explorer in v1.7 that is around the corner ... 
https://github.com/rootzoll/raspiblitz/issues/2096

nyxnor, 2021-04-18:
@pinnubi Can you try tge v1.7RC2 on the dev branch to see if this still occurs pls?

nyxnor, 2021-05-03:
@pinnubi Was this solved with v1.7.0?

closed 2021-05-16 by rootzoll