raspiblitz/raspiblitz #2145: HOW TO: Setup new raspiblitz with txindex=1, copy blockchain from (local net) full node with txindex=1 ?

david-bakin, 2021-04-05:
I see that in the past raspiblitz ran with txindex=1 but now it does not.

I'd like to run with txindex=1; and I'd like to initialize my (shortly-to-be-built) raspiblitz from a local server's bitcoin core blockchain that has txindex=1.

How do I do this?

nyxnor, 2021-04-07:
On the new version v1.7 (testing) will have the option to switch from 0 to 1.  
`sudo bash config.scripts/network.txindex.sh on`

david-bakin, 2021-04-07:
Great! I notice that on initializing the blockchain by copying it from a remote computer, via [`50copyHDD.sh` in `dev`](https://github.com/rootzoll/raspiblitz/blob/dev/home.admin/50copyHDD.sh), that script isn't going to copy over ./indexes (if present) from the remote.  So presumably the indexing would happen on starting bitcoind.  

If I modified the script to copy the indexes directory (or just did it manually via scp) before starting bitcoind the first time: would that work (i.e., to avoid bitcoind reindexing the blockchain)?  

(My questions is because I'm not sure if the indexes are bitcoind installation-specific or not.)

nyxnor, 2021-04-07:
Yes.
Fork Raspiblitz, modify the scripts, build your sd card with the changes from your fork, change the default txindex to 1.
You can do whatever you feel like.

david-bakin, 2021-04-07:
ok, thanks.

closed 2021-04-07 by david-bakin