raspiblitz/raspiblitz #2150: Touchscreen for 64bit drivers

rootzoll, 2021-04-07:
I merged the display refactoring. Switching on the touchscreen is working now again. But the calibration seems not working correctly yet. @openoms it would be great if you can take a look .. if we can get this fixed on the RC2 that would be great.

rootzoll, 2021-04-08:
We will take a look at the touchscreen calibration again after RC2 for now.

rootzoll, 2021-04-13:
@frennkie any luck so far to getting the calibration back on track?

rootzoll, 2021-04-15:
OK ... I got the touchscreen running again even with calibration and i2c-off :) Making some last test tomorrow, but looks good.

rootzoll, 2021-04-16:
@frennkie after upgrade from 1.6.3 to v1.7.0 (on my latest tests with the merged latest changes into dev) the touchscreen was working but needed a new calibration. After new calibration the touchscreen  restarted multiple times until it worked. This is the debug:

```
** Message: 01:05:33.299: app.vala:76: Launching unclutter
** Message: 01:05:33.308: app.vala:76: Launching xset
** Message: 01:05:33.312: app.vala:76: Launching xset
** Message: 01:05:33.316: app.vala:76: Launching xset
** Message: 01:05:33.322: app.vala:76: Launching sh
** Message: 01:05:33.330: options.vala:107: Create build-in Clipboard
** Message: 01:05:33.332: main.vala:441: Check keymap_mode (null)
** Message: 01:05:33.339: app.vala:76: Launching /usr/bin/ssh-agent
SSH_AUTH_SOCK=/tmp/ssh-verko5jOI2MK/agent.10267; export SSH_AUTH_SOCK;
SSH_AGENT_PID=10272; export SSH_AGENT_PID;
echo Agent pid 10272;
** Message: 01:05:33.499: app.vala:130: xset exit with this type of exit: 0
** Message: 01:05:33.500: app.vala:139: Exit normal, don't reload
** Message: 01:05:33.500: app.vala:130: xset exit with this type of exit: 0
** Message: 01:05:33.500: app.vala:139: Exit normal, don't reload
** Message: 01:05:33.500: app.vala:130: xset exit with this type of exit: 0
** Message: 01:05:33.500: app.vala:139: Exit normal, don't reload
** Message: 01:05:33.500: app.vala:130: /usr/bin/ssh-agent exit with this type of exit: 0
Openbox-Message: Unable to find a valid menu file "/usr/share/lxde/openbox/menu.xml"
```

So it seems the `/usr/share/lxde/openbox/menu.xml` was not ready. I think this is not critical, but can you give me a hint how to improve on that for v1.7.1?

rootzoll, 2021-04-16:
found a small bug to be fixed before final: when recovering with touchscreen on it ended up with `hdmi_hotplug=1` still in `/boot/config.txt` which gave a black screen (was no able to see the final recovery info). Check why. Maybe get rif of hdmi-hotplug completly for now if it just makes trouble.

rootzoll, 2021-04-19:
OK Touchscreen is now looking good .. even Calibration is working. Closing issue for v1.7 release.

closed 2021-04-19 by rootzoll

LISADONALDS090, 2022-05-19:
HAVE YOU BEEN SCAMMED OF YOUR COINS AND WANTS TO GET IT RECOVERED BACK???
I invested a total of$95,000.00USD worth of Bitcoin with an online company who does trading and was guaranteed a payout of 25% a week. They ended up shutting down their company website but their website was still running. So I could still see my dashboard at that time and instead of paying weekly they ended up compounding my money. It ended up compounding to$179, 000USD so I requested for a withdrawal which was declined before they shutdown their website, I complained to my colleague at work who directed me about this recovery expert, Mr. Morris Ray, that helped him, I contacted him immediately, what surprised me most, was that I recovered my money that same week. You can contact him on his email at MorrisGray830 @ Gmail . com and on WhatsApp: + 1 (607) 698-0239 and he will assist you on the steps to recover your invested funds. Just trying to help those who where scammed just the way I was too.