raspiblitz/raspiblitz #2157: Fresh 1.7RC1 still fails to HTTPS

The34Goon, 2021-04-08:
LIT in Chrome
ERR_CONNECTION_TIMED_OUT

LIT in Tor
Details: 0xF2 — Introduction failed, which means that the descriptor was found but the service is no longer connected to the introduction point. 
It is likely that the service has changed its descriptor or that it is not running.

LIT in Blitz SSH Menu:
Can't open /home/lit/.lit/tls.cert for reading, No such file or directory
547891533696:error:02001002:system library:fopen:No such file or directory:../crypto/bio/bss_file.c:69:fopen('/home/lit/.lit/tls.cert','r')
547891533696:error:2006D080:BIO routines:BIO_new_file:no such file:../crypto/bio/bss_file.c:76:
unable to load certificate

RTL/Thunderhub in Chrome HTTPS

Can't accept self-signed cert:
(after: export to file, add to trusted source, import and restart)
This server could not prove that it is 192.168.0.XXX; its security certificate is from localhost.
NET::ERR_CERT_COMMON_NAME_INVALID

RTL/Thunderhub in Tor
super slow.

Debug:
https://termbin.com/0ujs

The34Goon, 2021-04-09:
OK.

So RC2 will fix the CERT for HTTPS?

rootzoll, 2021-04-09:
This looks like something went wrong during the RC1 install, maybe related to the fs expand error. I just tested the Thunderhub & LIT again on the RC2 .. there is no problem with the certs with my test machine. I am preparing the RC2 at the moment  wil be available over the weekend. Please give it another test run and report any further error.

The34Goon, 2021-04-10:
More than happy to test it again.

The day-1-IP shown in the cert wasn't used in months.
I don't know where the Raspiblitz fetched it from..?

I'll try with the "old" ...144 IP fixed from router and regenerate tls from ssh menu.
[image]

rootzoll, 2021-04-10:
@frennkie can you take a look?

frennkie, 2021-04-10:
This is using the self signed certificate created by LND, right?
Which Browser (with version) are you using?

> So RC2 will fix the CERT for HTTPS?

What is your expectation here? A self signed cert will not be accepted by common browsers without further actions. What does work with a default browser is a Let's Encrypt certificate ...

The34Goon, 2021-04-10:
_This is using the self signed certificate created by LND, right?_

-Yes

_Which Browser (with version) are you using?_

-Brave, Chrome, Firefox, Tor -all latest version

_What is your expectation here?_ 
-To connect to my apps over HTTPS, as suggested in Raspiblitz SSH menu and get rid of the security warnings.
"Open in your local web browser & accept self-signed cert"

-Also I'd like to know why the cert contains the day1 IP entry after flashing fresh SD 

_A self signed cert will not be accepted by common browsers without further actions._ 

-aka. Import Blitz to trusted source? 
which worked fine in the past but does not since flashing fresh SD

rootzoll, 2021-04-10:
Here is the RC2 for download & test: https://github.com/rootzoll/raspiblitz/tree/dev#downloading-the-software

rootzoll, 2021-04-10:
As a take-away from a in depth chat I think its best to have the first link shown on additional Apps should be a simple `https` not the `https` so that newbies dont get scared and confused when calling it in the local browser. All other fullnodes do this also and we can still give the https & tor address below with a "for more security".

With v1.8 this will change for newbies anyway - because we will go with Tor browser by default.

frennkie, 2021-04-10:
I assume this is a decision based on the assumption that this is a connection on a local LAN (at home) - right? 
Where a "man-in-the-middle" would most likely already mean that the computer of the user is compromised.

Because in a scenario where the connection from the user to the RaspiBlitz travels over an a network (e.g. "Internet") I think it's absolute not acceptable to transmit credentials that allow access (e.g. to funds) in clear text. We also don't use telnet instead of ssh because there is a "weird fingerprint" message on the initial connection.

The34Goon, 2021-04-11:
The SSH menu asks you to accept the self-signed certificate, which is no longer possible.
Dev is telling you to just ignore the fat, red warning in the local browser, it's meaningless.
The warning tells you, "Attackers may be trying to steal your information."

I would say there is room for improvement here, if you really want to move noobs off Coinbase.
Either link to the FAQ and explain what's going on or ditch the message and only show Tor option.

rootzoll, 2021-04-18:
Because LIT is using its own local cert and RTL, Thunderhub, etc is using the RaspBlitz managed one. There will be no consistent solution for all additional apps in v1.7 I pushed some to v1.7.1 wth this issue #2194 ... but I started to change some that were easy to give the users options:

So for example RTL is now showing this:
[image]

@frennkie I think for now its fair to assume that a RaspiBlitz is running within the local network behind NAT and so offering a unsecure HTTP is OK as an option should be OK ... because normal/new users seeing a WARNING in their browser after installing an app will just assume its broken. For v1.8 we target to launch the WebUI for new users - that will use TorBrowser by default and so we will link those additional service over onion-adresses anyway.

rootzoll, 2021-04-19:
OK this looks good for v1.7 .. closing issue.

closed 2021-04-19 by rootzoll