raspiblitz/raspiblitz #2162: New: Homer Dashboard

rootzoll, 2021-04-08:
see https://github.com/rootzoll/raspiblitz/pull/2152/#issuecomment-815985212

rootzoll, 2021-04-08:
OK I did some small fixes, put it as part of the recovery, added the `update` and was working on persitence for the config.

So the config will be stored and should be edited by the user at:`/mnt/hdd/app-data/homer/config.yml` ... this file will be lionked into the homer nginx assets in the end. Because it in `app-data` it will survive updates & migrations

rootzoll, 2021-04-08:
@hashes4merkle if we can get this stable - happy to add to the final v1.7 release (targeted for next week)

hashes4merkle, 2021-04-12:
@rootzoll Tested install, update and delete on a new environment. Looks good.  

Still waiting on comments from upstream to see if those dependency issues can be resolved.  Would like to avoid using --legacy-peer-deps if we can.

rootzoll, 2021-04-13:
OK so maybe lets push ist to v1.7.1

hashes4merkle, 2021-04-22:
Homer dependencies resolved.  We can pull from master branch or wait for the next release.

rootzoll, 2021-04-23:
Sweet :) V1.7 final is quite close .. lets put it into v1.7.1

rootzoll, 2022-05-17:
Merged for v1.8.0 release - final testing.