raspiblitz/raspiblitz #2163: npm audit reveals vulnerablities

kamordi, 2021-04-08:
While installing mempool I noticed the output warned about 18 vulnerabilities. I checked this via 

`admin@raspberrypi:/home/mempool/mempool/frontend $ npm audit`

[...] [List of vulnerablities]

`found 18 vulnerabilities (4 moderate, 14 high) in 1723 scanned packages
  run npm audit fix to fix 18 of them.`

It is caused by the package  [y18n](https://www.npmjs.com/advisories/1654). I checked this issue in the [mempool-repo](https://github.com/mempool/mempool/issues?page=4&q=is%3Aissue+is%3Aopen), but this wasn't even mentioned there. Should that cause concern, or is it already fixed in next milestones?

rootzoll, 2021-04-08:
@wiz can you comment in this?

wiz, 2021-04-09:
npm deps are already updated in upcoming v2.2 release, if user is concerned they can change tag to master

kamordi, 2021-04-16:
I was thinking about to provide a kind of script, which can test for this kind of vulnerabilities, shown by `npm audit`. So this should be executed, maybe once a month, for all installed programs via npm, not only mempool. Would this be a benefit or would the issues even be fixed by continuous upgrading the installations itself?

wiz, 2021-04-16:
Ever since GitHub acquired npm the repo maintainers get notifications from GitHub all the time about npm dependency vulnerabilities, so I think if you fork mempool into your own GitHub account you can get the same notifications?

kamordi, 2021-04-16:
To avoid misunderstandings.

I'am in the view of a runner of the node. To make it clear, a not professional/experienced runner. I saw the warnings and fixed them by running `npm audit fix`.  I don't know if this was the correct way. I did it in a understanding of kind of responsibility for my node. So when I don't check and don't run every new release I see a risk for the runner. From my point of experience, I can't imagine if this risk would be significant. 

So if there is no need to fix this warnings by the user, I'm fine with it.

wiz, 2021-04-16:
The main problem is that RaspiBlitz hasn't shipped a new version of mempool in a long time, so once they do I think those warnings will go away.

kamordi, 2021-04-16:
I don't want to blame mempool. When I noticed it, I checked it on other installations (RTL, RPC), too. I received also warnings, which I fixed with `npm audit fix`. The question is for me, if the user should have the responsibility, which would be ok, but then he should be aware of and should have the chance to react to this issues.

rootzoll, 2021-04-16:
v1.7RC2 is available to test here with mempool v2.1.2: https://github.com/rootzoll/raspiblitz/blob/dev/README.md#downloading-the-software

And a final v1.7 should now be out within a week .. @wiz sorry that it took so long, but it was a major update for RaspiBlitz .. the next one v1.7.1 will be faster and then can come with v2.2 of mempool.

rootzoll, 2021-08-24:
I said `v1.7.1 will be faster` and then we decided to add c-lightning and make a big refactor of the setup process and schwups it August - LOL 

But for the v1.7.1rc1 will already contain mempool v2.2.1 - so setting this issue to final testing.

rootzoll, 2021-10-10:
mempool v2.2.2 tested and will be part of v1.7.1 final release.

closed 2021-10-10 by rootzoll

flyerbear, 2022-05-01:
I am running v1.7.2 and see similar warnings during mempool space and Thunder Hub installs. Should I attempt to fix these with npm audit fix or will this break upstream dependencies?

Also, should I upgrade npm from 7.24.0 to 8.7.0 or will this cause problems?

Thanks!