raspiblitz/raspiblitz #2164: LIT + RTL .. make them work together

rootzoll, 2021-04-08:
doenst need to be part of RC2 ... but should be part of the final release.

openoms, 2021-04-09:
Covered in: https://github.com/rootzoll/raspiblitz/commit/be6e9bbbc9440b1954c30c74a0fd28ab24555ea0 and
https://github.com/rootzoll/raspiblitz/commit/172bb54e6d895ee6b173650961572854bc64cf0f
commited directly since were small changes and tested ok quickly.

rootzoll, 2021-04-19:
Looks good sofar ... Final Test: Upgrade my production node to v1.7RC3 and test LoopIn with RTL.

openoms, 2021-04-19:
Bad news on this, because the Loop REST interface is not actullay implemented in Lit yet: https://github.com/Ride-The-Lightning/RTL/issues/660

Do it means Loop will be only usable with LiT and the CLI, not RTL in this version.

rootzoll, 2021-04-19:
Thanks for the clarification ... I will push this issue to the v1.7.1

rootzoll, 2022-05-09:
LIT will not yet be part of WebUI v1.8.0 release ... but pushing this to v1.8.1 to recheck LIT & RTL integration again.

rootzoll, 2022-12-13:
LIT also not yet part of 1.9 ..m pushing it to v1.10

closed 2023-11-14 by rootzoll