raspiblitz/raspiblitz #2167: RaspiBlitz v1.7 final Release

rootzoll, 2021-04-09:
- [ ] create v1.7 branch
- [ ] set v1.7 on sd card 
- [ ] update latest code on sd card
- [ ] make a final v1.7 release image
- [ ] sign & update md5 of image

closed 2021-04-25 by rootzoll

P8GRP, 2021-04-25:
Congrats! Thanks more importantly! Loading it on my second Virgin node now.

P8GRP, 2021-04-26:
Started new RaspiBlitz load after scrubbing 1.6.3 image that was on it. Flashed 1.7 onto the micro sd. The bootup appears to be stalled at 
[image]

P8GRP, 2021-04-26:
I rebooted the start up however it comes back to this point and fails and repeats. Any assistance is appreciated.

openoms, 2021-04-26:
@P8GRP  Reflash the SDcard with the verified image. Ideally use a new, quality card.

P8GRP, 2021-04-28:
Reflashed a new SDcard. Same issue. I cannot resolve this? installing bash completion for bitcoin-cli and lncli--2021-04-28 20:44:10--  https://raw.githubusercontent.com/bitcoin/bitcoin/master/contrib/bitcoin-cli.bash-completionResolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... failed: Connection timed out.Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... failed: Connection timed out.