raspiblitz/raspiblitz #2173: Integrate Specter Desktop in new Connect menu

dennisreimann, 2021-04-12:
Having a guide about [how to connect Specter Desktop with the RaspiBlitz](https://d11n.net/connect-specter-desktor-with-raspiblitz.html#on-the-raspiblitz) is good – obsoleting it with software would be even better. 

As the RaspiBlitz v1.7.0 will feature a new Connect menu, it would be good to integrate Specter Desktop there: The user can opt-in for making the bitcoind RPC available on the local network and a script could preconfigure the settings and firewall according to the guide. It could then restart bitcoind and show the user a confirmation and information on how to configure the connection in Specter Desktop.

openoms, 2021-04-12:
Good news, you can try this in the dev branch already: https://github.com/rootzoll/raspiblitz/pull/2119

openoms, 2021-04-12:
[image]
[image]
As testing now the workflow can be improved a it exits in the main menu after activatiing, but here is the option to show the details:
[image]
```
192.168.3.161
RPC username:
raspibolt

RPC password:
REDACTED

Host on the local network (make sure to connect from the same network):
192.168.3.161

Port:
8332

More documentation at:
https://github.com/openoms/joininbox/blob/master/prepare_remote_node.md
```

dennisreimann, 2021-04-12:
Awesome, so this will be part of v1.7? Then I can update the article and point to that feature instead. Great work! 💪

nyxnor, 2021-04-12:
@openoms you are so fast, thanks for the great work.

rootzoll, 2021-04-18:
Hmm OK .. I fixed some small problem with the port ... but on my tests for v1.7 it does not look as the connection with Specter Desktop App is sucessfull - see:

[image]

I will have the script in v1.7 (because its a generic RPC connection dialog anyway) but push this to v1.7.1 ... for finalization. @openoms if you have a quick fix for final v1.7 I am happy to add.

rootzoll, 2021-04-18:
OK .. the raspiBlitz needed a reboot, then the connection was working with all checkmarks. Maybe this was due to the activation of the bitcoin wallet. So lets finalize this in v1.7.1

rootzoll, 2021-10-10:
[image]
All checkmarks green - closing issue for v1.71. release.

closed 2021-10-10 by rootzoll