raspiblitz/raspiblitz #2176: 99% Sync failing -- Problem Bootstraping

ag2121-ag, 2021-04-13:
My pi seems stuck at 99% and I ran the following and get

sudo grep -i tor /var/log/syslog | nc termbin.com 9999

```
[Apr 13 13:00:03 raspberrypi systemd[1]: /lib/systemd/system/fail2ban.service:12: PIDFile= references path below legacy directory /var/run/, updating /var/run/fail2ban/fail2ban.pid → /run/fail2ban/fail2ban.pid; please update the unit file accordingly.
Apr 13 13:00:04 raspberrypi systemd[1]: Started RaspiBlitz Background Monitoring Service.
Apr 13 13:00:04 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: peers=
Apr 13 13:00:11 raspberrypi bonus.sphinxrelay.sh[5521]: # ok - copied fresh config.json & app.json into dist directory
Apr 13 13:00:51 raspberrypi tor[681]: Apr 13 13:00:51.000 [notice] New control connection opened from 127.0.0.1.
Apr 13 13:01:11 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: peers=0
Apr 13 13:01:11 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: ZERO PEERS DETECTED .. doing out-of-band kickstart
Apr 13 13:01:11 raspberrypi _background.sh[5442]: #network.monitor.sh peer-kickstart
Apr 13 13:01:13 raspberrypi _background.sh[5442]: addressFormat='tor'
Apr 13 13:01:16 raspberrypi npm[5517]: 2021-04-13T12:01:16.691Z btcexp:app Loading historical data for chain=main
Apr 13 13:01:53 raspberrypi tor[681]: Apr 13 13:01:53.000 [notice] New control connection opened from 127.0.0.1.
Apr 13 13:02:55 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: peers=0
Apr 13 13:02:55 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: ZERO PEERS DETECTED .. doing out-of-band kickstart
Apr 13 13:02:55 raspberrypi _background.sh[5442]: #network.monitor.sh peer-kickstart
Apr 13 13:02:57 raspberrypi _background.sh[5442]: addressFormat='tor'
Apr 13 13:03:12 raspberrypi tor[681]: Apr 13 13:03:12.000 [notice] We'd like to launch a circuit to handle a connection, but we already have 32 general-purpose client circuits pending. Waiting until some finish.
Apr 13 13:04:50 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: peers=0
Apr 13 13:04:50 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: ZERO PEERS DETECTED .. doing out-of-band kickstart
Apr 13 13:04:50 raspberrypi _background.sh[5442]: #network.monitor.sh peer-kickstart
Apr 13 13:04:52 raspberrypi _background.sh[5442]: addressFormat='tor'
Apr 13 13:06:34 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: peers=0
Apr 13 13:06:34 raspberrypi _background.sh[5442]: Blockchain Sync Monitoring: ZERO PEERS DETECTED .. doing out-of-band kickstart
Apr 13 13:06:34 raspberrypi _background.sh[5442]: #network.monitor.sh peer-kickstart
Apr 13 13:06:36 raspberrypi _background.sh[5442]: addressFormat='tor'
Apr 13 13:07:14 raspberrypi tor[681]: Apr 13 13:07:14.000 [warn] Problem bootstrapping. Stuck at 95% (circuit_create): Establishing a Tor circuit. (DONE; DONE; count 10; recommendation warn; host 03910F285A33F365838EC66EF2C2EF754D046760 at 148.251.237.219:443)
Apr 13 13:07:14 raspberrypi tor[681]: Apr 13 13:07:14.000 [warn] 10 connections have failed:
Apr 13 13:07:14 raspberrypi tor[681]: Apr 13 13:07:14.000 [warn]  5 connections died in state handshaking (Tor, v3 handshake) with SSL state SSL negotiation finished successfully in OPEN
Apr 13 13:07:14 raspberrypi tor[681]: Apr 13 13:07:14.000 [warn]  3 connections died in state handshaking (TLS) with SSL state SSLv3/TLS write client hello in HANDSHAKE
Apr 13 13:07:14 raspberrypi tor[681]: Apr 13 13:07:14.000 [warn]  1 connections died in state handshaking (TLS) with SSL state SSLv3/TLS write finished in HANDSHAKE
Apr 13 13:07:14 raspberrypi tor[681]: Apr 13 13:07:14.000 [warn]  1 connections died in state connect()ing with SSL state (No SSL object)
Apr 13 13:07:26 raspberrypi tor[681]: Apr 13 13:07:26.000 [warn] Problem bootstrapping. Stuck at 95% (circuit_create): Establishing a Tor circuit. (DONE; DONE; count 11; recommendation warn; host 5790B62AD6C79A1092B30993A8F0F7F23A969FD6 at 51.83.237.59:9001)
Apr 13 13:07:26 raspberrypi tor[681]: Apr 13 13:07:26.000 [warn] 11 connections have failed:
Apr 13 13:07:26 raspberrypi tor[681]: Apr 13 13:07:26.000 [warn]  5 connections died in state handshaking (Tor, v3 handshake) with SSL state SSL negotiation finished successfully in OPEN
Apr 13 13:07:26 raspberrypi tor[681]: Apr 13 13:07:26.000 [warn]  3 connections died in state handshaking (TLS) with SSL state SSLv3/TLS write client hello in HANDSHAKE
Apr 13 13:07:26 raspberrypi tor[681]: Apr 13 13:07:26.000 [warn]  2 connections died in state handshaking (TLS) with SSL state SSLv3/TLS write finished in HANDSHAKE
Apr 13 13:07:26 raspberrypi tor[681]: Apr 13 13:07:26.000 [warn]  1 connections died in state connect()ing with SSL state (No SSL object)
Apr 13 13:07:34 raspberrypi tor[681]: Apr 13 13:07:34.000 [warn] Problem bootstrapping. Stuck at 95% (circuit_create): Establishing a Tor circuit. (DONE; DONE; count 12; recommendation warn; host 9DA7EF9E9D6070CD8B7059B8A8CC5B41CEC02726 at 192.99.151.170:9001)
Apr 13 13:07:34 raspberrypi tor[681]: Apr 13 13:07:34.000 [warn] 12 connections have failed:
Apr 13 13:07:34 raspberrypi tor[681]: Apr 13 13:07:34.000 [warn]  6 connections died in state handshaking (Tor, v3 handshake) with SSL state SSL negotiation finished successfully in OPEN
Apr 13 13:07:34 raspberrypi tor[681]: Apr 13 13:07:34.000 [warn]  3 connections died in state handshaking (TLS) with SSL state SSLv3/TLS write client hello in HANDSHAKE
Apr 13 13:07:34 raspberrypi tor[681]: Apr 13 13:07:34.000 [warn]  2 connections died in state handshaking (TLS) with SSL state SSLv3/TLS write finished in HANDSHAKE](url)
```

It seems to be a problem with TOR but I am unsure how to correct this. Thanks

nyxnor, 2021-04-13:
The logs: `Directory /mnt/hdd/tor/web80/ cannot be read: Permission denied`
The correction: https://github.com/rootzoll/raspiblitz/blob/da986f8c9ab40575d44b627ad6f39c6ccdcc7a68/home.admin/config.scripts/internet.tor.sh#L301
```
sudo chmod -R 700 /mnt/hdd/tor
sudo chown -R debian-tor:debian-tor /mnt/hdd/tor
sudo systemctl restart tor@default
sudo journalctl -eu tor@default
```

nyxnor, 2021-04-18:
@ag2121-ag Please close this issue if this is solved.

ag2121-ag, 2021-04-18:
This worked! thanks for the help!

closed 2021-04-18 by ag2121-ag