raspiblitz/raspiblitz #2180: Tor everything

nyxnor, 2021-04-16:
Yes, everything. `torsocks` when reaching outside, downloading electrum per example. Openoms suggested `iptables` to make all traffic go through Tor. That would facilitate the process. Does not take much time to download, as the packages Blitz download are very light even in the bonus scripts.
Even torify the the debug logs link, Im trying onionshare, but debian version is too old, need to build from source.

rootzoll, 2021-04-16:
Definitely something to look into and test for next major release - added to milestone 1.8

rootzoll, 2021-11-09:
See next step here: https://github.com/rootzoll/raspiblitz/pull/2672#issuecomment-964126808

rootzoll, 2021-12-11:
OK basic refactor of Tor install is merged for v1.7.2 ... but more work is needed to make "Tor everything". Also I modified the tor install script now that basic Tor can be installed already during sd card build when there is no HDD/SSD yet. Basically the tor@default service is working after sd card build but disabled so far that if the final user boots up a fresh raspiblitz that tor will not try to connect to the tor network and my reveal the user using tor to the ISP. With a fresh sd card to setup a RaspiBlitz the user should have the option to configure bridges and similar first before tor is enabled (to be discussed further)

For the v1.7.2 by default after the setup menu before setup/provisioning the `tor.install.sh enable` is then called which will either create the needed data directories on the now available HDD/SSD or will use the existing data there in a recovery scenario. 

So for the v1.8.0 either during setup dialogs or as part of the [Basic Setup Menu](https://github.com/rootzoll/raspiblitz#basic-setup) the user could configure the Tor connection.

nyxnor, 2021-12-11:
Bridges packages are being built but there is not easy menu yet, it will be imported from torbox on a later release.

nyxnor, 2021-12-22:
Instead of preprending every `curl` and `wget` we can use environment variables.
Easiest to `export` them on a file sourced every boot or a one time per script with `env`
### curl
```sh
env http_proxy=socks5h://127.0.0.1:9050 HTTPS_PROXY=socks5h://localhost:9050 ALL_PROXY=socks5h://127.0.0.1:9050
```
or in `~/.curlrc`:
```
http_proxy=socks5h://127.0.0.1:9050
HTTPS_PROXY=socks5h://127.0.0.1:9050
ALL_PROXY=socks5h://127.0.0.1:9050
```
or we can set the `CURL_HOME` to `/etc/curlrc` or specify this file with `-K, --config FILE` every time.

### wget
in the `/etc/wgetrc` for system wide or `~/.wgetrc` for user only but I prefer system wide
```sh
use_proxy=yes
http_proxy=127.0.0.1:9050
https_proxy=127.0.0.1:9050
```

### git
we can set the global proxy:
```sh
git config --global http.proxy 'socks5://127.0.0.1:9050' 
```
or one time per clone
```sh
git -c http.proxy=socks5h://127.0.0.1:9050 clone http://nxshomzlgqmwfwhcnyvbznyrybh3gotlfgis7wkv7iur2yj2rarlhiad.onion/git/bitcoin.git
```

### apt
depends on the repo but Debian and Tor Project have an onion repo, so that is all wee need:
install `apt-transport-tor`
on `comment default values on `/etc/apt/sources.list`
include on `/etc/apt/sources.list.d/debian.list`
```
deb  tor+http://2s4yqjx5ul6okpp3f2gaunr2syex5jgbfpfvhxxbbjwnrsvbk5v3qbid.onion/debian          bullseye            main
deb  tor+http://2s4yqjx5ul6okpp3f2gaunr2syex5jgbfpfvhxxbbjwnrsvbk5v3qbid.onion/debian          bullseye-updates    main
deb  tor+http://5ajw6aqf3ep7sijnscdzw77t7xq4xjpsy335yb2wiwgouo7yfxtjlmid.onion/debian-security bullseye/updates    main

#deb tor+http://2s4yqjx5ul6okpp3f2gaunr2syex5jgbfpfvhxxbbjwnrsvbk5v3qbid.onion/debian          bullseye-backports  main
```
include on include on `/etc/apt/sources.list.d/torproject.list`
```
deb [arch=arm64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye main
deb-src [arch=arm64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org  bullseye main
```