raspiblitz/raspiblitz #2183: UPDATE option for Bitcoin Core

openoms, 2021-04-16:
Given the coming softwork signaling and activation process there should be an easy option to update the Bitcoin Core version manually.

Alternative clients would likely to break things so wouldn't include switching in the menu.

There are these scripts with automatic PGP verification which to be adapted for the menu:
https://github.com/openoms/bitcoin-tutorials/tree/master/raspiblitz.updates#bitcoin-core-updates

rootzoll, 2021-04-20:
Tested with update to v0.21.1rc1 ... worked :) Very sweet - will be part of v1.7 and can be further tested already in v1.7RC3. Closing issue.

Maybe adding a reboot in the end would make sense for the future - because my LND restarted/locked and I was offered an option to unlock. Opened an issue for v1.7.1 for liitle UX improvements. #2211

openoms, 2021-04-20:
I think reboots are way too heavy handed and adds unnecessary downtime. I would rather see work minimizing the need for full restarts.

rootzoll, 2021-04-20:
I agree that reboots are heavy - but they are an effective way to get to a clean state after a change like that again. My LND restarted while bitcoin core got updated, my wallet was locked (so downtime also) and there was no unlock screen. From the UX kind of perspective my node was in a unclean state.

openoms, 2021-04-21:
Yes, LND restarts when the bitcoin RPC interface is not accessible after a timeout (10s by default): https://github.com/lightningnetwork/lnd/blob/1a8d1874a545c456744ad26d0d3a64b9bed58f0e/sample-lnd.conf#L871

So LND is stopped and restarted preemptively by the bitcoin core update script.

Made a note on the end of the script to use `lncli unlock`, but agree that the usual lnd unlock script should be triggered. No need for full restart.

openoms, 2021-04-21:
On an other related note here is great step by step guide to install and alternative client (like the UASF on from Luke Dashjr) manually which seems like it can work for the RaspiBlitz also: https://github.com/Stadicus/RaspiBolt/issues/722

Not planning to oinculde it in the menu since alternative implementations can cause other issues which we have no resources to test against.

rootzoll, 2021-04-23:
This tested good now with LND unlocking. Thx. Closing issue for v1.7 final relase.

closed 2021-04-23 by rootzoll