raspiblitz/raspiblitz #2184: Move repo to Organization to give Triage access

rootzoll, 2021-04-16:
To be able to give issue managing access right to the repo I would need to move it into an organization - see:
https://github.com/rootzoll/raspiblitz/issues/193#issuecomment-821088073

This is to research if its possible without breaking links, etc for the v1.7.1

openoms, 2021-04-16:
We are squatting https://github.com/raspiblitz already.
Will look around what more is possible here: https://github.com/orgs/raspiblitz/dashboard

nyxnor, 2021-04-16:
I dont want to have privileged options as Im still learning. Triage seems good enough if I can close old issues with it https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-permission-levels-for-an-organization

rootzoll, 2021-09-09:
I move this to v1.8 - just to get v1.7.1 quicker ready.

Also as a sidenote - dispite gossip I heard ... it looks like the repo can keep its followers & stars when getting moved:
https://docs.github.com/en/github/administering-a-repository/managing-repository-settings/transferring-a-repository

rootzoll, 2022-05-09:
Its still on the todo list ... just gets more complex now with the new repos for BlitzAPI & WebUI. Maybe thats something for a community event end of year to do together. Will push it up to v1.8.2 for now.

cstenglein, 2023-06-07:
Done :raised_hands:

closed 2023-06-07 by cstenglein