raspiblitz/raspiblitz #2185: Hardware test is not ruuning on 64bit Raspberry OS

openoms, 2021-04-16:
Just trying the `HARDWARE` test menu on 1.7RC2
```
*** Hardware Test Report ***

the OS is not Raspbian - the stresstest is only for the Raspberry Pi

/home/admin/05hardwareTest.sh: line 26: /home/admin/stresstest.report: No such file or directory
/home/admin/05hardwareTest.sh: line 34: [: -gt: unary operator expected
/home/admin/05hardwareTest.sh: line 112: [: -gt: unary operator expected
```

Not running during the first setup either.

Question @rootzoll if we should fix this or just remove the option?
 It was developed for the RPi3 and  not sure if is very useful for the RPi4.

nyxnor, 2021-04-17:
Openoms, this month there was a wave of hardware issues on the debug logs shared on telegram. I think this is best to be reworked. Maybe just deleting this part as there are many more OS available now https://github.com/rootzoll/raspiblitz/blob/c6c8989f01ad66233f6c25f0aa89411300617beb/home.admin/config.scripts/blitz.stresstest.sh#L10

openoms, 2021-04-17:
The issues are often USB adapter related which this test does not cover and it has other mitigation in v1.7.

Voltage instability was an problem for the Rpi3, but not the rpi4.
Heat is of limited significance also and needs longer time to build up.

I don't feel the hardware test in this form is very useful and takes time to run. 
It is an easy fix to make it work, but I'd rather deactivate it for now.

rootzoll, 2021-04-17:
I agree - reducing complexity makes sense. I will remove hardware test for the v1.7

rootzoll, 2021-04-17:
OK removed the hardware test for start/setup, menu and debug. Now final testing if all still runs smoothly.

rootzoll, 2021-04-19:
Looks good - closing issue.

closed 2021-04-19 by rootzoll

maxgoldberg, 2021-04-30:
Changed the power supply today on my raspiblitz and got a message on the info screen about how I had an undervoltage report and I should run the hardware test, which appears to have been removed.

rootzoll, 2021-05-18:
@maxgoldberg thanks for the report (thats an old text) .. added a note to check this for v1.7.1 release- see: #2188