raspiblitz/raspiblitz #2188: Add minimal placeholder for Hardware Test again

rootzoll, 2021-04-17:
For v1.7 we removed the old hardware test because it was not much needed anymore when most run RP4 
see: https://github.com/rootzoll/raspiblitz/issues/2185#issuecomment-821759248

But it can make sense for little things like checking if HDD is connected to USB3 (not USB2) or check sd card speed, etc and maybe bigger things in the future. So in preperation for the v1.8 refactoring we should have a small placeholder for a hardware check on startup/setup and one in the repair/settings menus. Also to be part of the debug log so that on support people can spot problems with the hardware.

So v1.7.1 should introduce a little placeholder for the hardware test again - it can be further improved then even after v1.8

nyxnor, 2021-04-18:
- [ ] Mem test https://github.com/rootzoll/raspiblitz/issues/517

rootzoll, 2021-05-18:
see: https://github.com/rootzoll/raspiblitz/issues/2185#issuecomment-830345792

rootzoll, 2021-05-19:
use new script `blitz.hardware.sh` for this - see branch `1126-provision`

rootzoll, 2021-10-05:
SSD Hardwaretest see: #2599