raspiblitz/raspiblitz #2191: Slow to sync, odd bitcoind behavior.

benjamin-wilson, 2021-04-17:
I have all the hardware that was suggested however I'm 4 days into the sync and only passed 50%. It doesn't seem to be utilizing system resources much. I've documented my findings on stack exchange but posting here for more specific raspiblitz help.

https://bitcoin.stackexchange.com/questions/105493/raspiblitz-slow-sync-performance

nyxnor, 2021-04-17:
The percentage is based on the (current block downloaded)/(total blocks). The first blocks were not big, most recent blocks start getting full, that is why it slow down the percentage number. Also, to print debug logs:
 `/home/admin/XXdebugLogs.sh | torsocks nc termbin.com 9999`.
 
 If it was presented how much data in bytes were downloaded, and how many remains, it would be clear the example I gave above, PRs welcome.

benjamin-wilson, 2021-04-17:
It's not a matter of newer or older blocks or the sync progress. Simply restarting bitcoind will 10x the sync speed for a short while. I go over this in detail on stackexchange. Currently the node is behaving and syncing at an acceptable rate but if it slows down again I'll post XXdebugLogs.

benjamin-wilson, 2021-04-18:
Woke up this morning and it's slow again. Logs: https://termbin.com/3ipr

nyxnor, 2021-04-18:
For what is worth, I can say this is normal behavior. The logs also seems fine. On start up, there is always a peak.

SCH00N3R, 2022-03-02:
I am seeing the same behaviour on my unit. Is it normal or consistent anomalous behaviour (in terms of what is expected)?
I can sync much faster by periodically rebooting the raspiblitz. As there is nothing about the data or peer connections that changes, it seems that it is likely that the process that indexes the blockchain data is suffering from some sort of bug that causes it to slow down. Something is cleared out by a reboot, and the process is able to resume at a normal rate for an hour or two and then begins to degrade again.

rootzoll, 2022-03-08:
Whats the temperature of your devices? ... the raspberrypi throttles down performance if it runs too hot, what might low down of sync extremly.

rootzoll, 2022-03-09:
Yeah HDDs are not recommended because their performace on initial sync is really bad and they are prune to lose power and data corrupt over the longtime - SSD should be the standard for running a RaspiBlitz if possible. Another reason why even a SSD can be slow is also, when you connect it on a RaspberryPi thru the USB2 ports (black) and not as it should be on the USB3 ports (blue).

closed 2022-03-09 by rootzoll