raspiblitz/raspiblitz #2192: Chantools Status menu show LND xPub

rootzoll, 2021-04-17:
Multiple people are asking how to get the xPub of LND on-chain wallet (to use with BTCPayServer for example). As far I was able to research its not a standard function of LND to export this. But Olli gave a hint that its a function of chantools.

So as a quick helper we can show the xpub in the SSH menu/status of chantools (once installed over SERVICES). 

The derivation path of LND seems to be `m/1017'/0'/6'/0/0` - see https://github.com/lightningnetwork/lnd/issues/732#issuecomment-565951975

openoms, 2021-04-17:
Be careful that the HD Root key is master PRIVATE key of the onchain part of the LND wallet.
about chantools: 
```
walletinfo          Shows info about an lnd wallet.db file and optionally extracts the BIP32 HD root key
```

For recovery scenarios there is the process described in: 
https://openoms.gitbook.io/lightning-node-management/technicals/restorelndonchainfundsinelectrum
but since it involves handling sensitive keys I would not recommend it for routine use.

Also funding when funding the wallet from electrum the UTXO mightn ot be picked up by LND without a rescan.
Thsi can lead to meany issues and would discourage this.

laurentdebricon, 2021-08-05:
i just tried : 
`sudo service lnd stop 
`
 because of : https://github.com/guggero/chantools/issues/18

`chantools walletinfo --walletdb ~/.lnd/data/chain/bitcoin/mainnet/wallet.db
` (you are asked to enter your C password to unlock the wallet)
you get the **xprv** and and also this output : 
```
Wallet  #scopes:

Scope:	m/49'/0'
  Number of internal np2wkh addresses:	0
  Number of external np2wkh addresses: 	0

Scope:	m/84'/0'
  Number of internal p2wkh addresses:	5
  Number of external p2wkh addresses: 	17

Scope:	m/1017'/0'
  Number of internal p2wkh addresses:	0
  Number of external p2wkh addresses: 	8
```

Then to get the onchain raspiblitz xpub : 
```
chantools derivekey --rootkey=your_xprv --path="m/84'/0'/0'"
```
and it works ! i can see, in a read only wallet for example, the correct funds of my raspiblitz on-chain wallet.

> @rootzoll "The derivation path of LND seems to be m/1017'/0'/6'/0/0"

this was not working, maybe we not talking of the same thing. :s

rootzoll, 2021-08-17:
This sounds like a good tool to add to the menu - but would need a bit of SSH-GUI and testing aorund it .. so I push this to v1.8.1 for now because of time prio. If some like to contribute a script or extend the `bonus.chantools.sh` for that feature earlier feel welcome.

openoms, 2022-05-18:
> lncli wallet accounts list

@guggero this is a very useful new function to create watch only wallets and construct PSBTs externally.
Only the question is if we still only need the `m/1017'/0'/6'` path (because `lncli wallet accounts list` shows the accounts :
```
            "derivation_path": "m/49'/0'/0'",
            "derivation_path": "m/84'/0'/0'",
            "derivation_path": "m/1017'/0'/0'",
         
            ... down to 
            
            "derivation_path": "m/1017'/0'/255'",

```
and a number of them has used addresses and even private keys. The internal wallet has 5 UTXOs currently and shows keys on these paths:
```
            "derivation_path": "m/49'/0'/0'",
            "external_key_count": 13,
            "internal_key_count": 0,
--
            "derivation_path": "m/84'/0'/0'",
            "external_key_count": 410,
            "internal_key_count": 60,
--
            "derivation_path": "m/1017'/0'/0'",
            "external_key_count": 309,
            "internal_key_count": 2,
--
            "derivation_path": "m/1017'/0'/1'",
            "external_key_count": 308,
            "internal_key_count": 0,
--
            "derivation_path": "m/1017'/0'/2'",
            "external_key_count": 308,
            "internal_key_count": 0,
--
            "derivation_path": "m/1017'/0'/3'",
            "external_key_count": 308,
            "internal_key_count": 0,
--
            "derivation_path": "m/1017'/0'/4'",
            "external_key_count": 308,
            "internal_key_count": 0,
--
            "derivation_path": "m/1017'/0'/5'",
            "external_key_count": 308,
            "internal_key_count": 0,
--
            "derivation_path": "m/1017'/0'/6'",
            "external_key_count": 0,
            "internal_key_count": 0,
```

guggero, 2022-05-18:
You only need the following paths:
 - `m/49'/0'/0'`: for nested segwit addresses (`lncli newaddress np2wkh`)
 - `m/84'/0'/0'`: for native segwit addresses (`lncli newaddress p2wkh`)
 - `m/86'/0'/0'`: for taproot addresses (`lncli newaddress p2tr`)

Everything below the `m/1017'/0'` is internal `lnd` stuff and there won't be any on-chain funds on those. The `m/1017'/0'/6'/0/0` key is the node's identity key. That can be useful to derive but it will never contain any funds.

openoms, 2022-05-18:
Thank you for the explanation, that's a very useful new feature!
It think we can expose the 3 pubkeys in the menu to make starting an external reqd only wallet easy and and offer an option to sign a PSBT.

guggero, 2022-05-18:
> has 3 transactions from 2019. One deposit and 2 channel opens (the second emptied the account with the change going to another account) - is this some deprecated behaviour of the LND onchain wallet?

I forgot to answer the last question. Am I correct in assuming that the three transactions were actually on the `m/1017'/0'/3'/0/X` path? That is where the funds go to if the remote node force closes. But `lnd` detects those funds and sweeps them to the "normal" on-chain wallet (`m/84'`). That is indeed some legacy behavior. Those keys used to be tweaked but since the `static_to_remote` feature was implemented a few years ago, they are normal keys that can directly be derived from the seed. So they could be keys under the default derivation path and we could avoid needing to sweep them. But we just didn't get to fixing that behavior.

openoms, 2022-05-18:
> Just leaving a hint here that with `lncli wallet accounts list` you can also get all `xpub`s of lnd's wallet.

`lncli wallet accounts list --name default`

shows only the accounts needed and no internals.

openoms, 2022-05-18:
To sign a PSBT:
```
# lncli wallet psbt finalize -h
NAME:
   lncli wallet psbt finalize - Finalize a Partially Signed Bitcoin Transaction (PSBT).

USAGE:
   lncli wallet psbt finalize [command options] funded_psbt

DESCRIPTION:
   
  The finalize command expects a partial transaction with all inputs
  and outputs fully declared and tries to sign all inputs that belong to
  the wallet. Lnd must be the last signer of the transaction. That means,
  if there are any unsigned non-witness inputs or inputs without UTXO
  information attached or inputs without witness data that do not belong
  to lnd's wallet, this method will fail. If no error is returned, the
  PSBT is ready to be extracted and the final TX within to be broadcast.

  This method does NOT publish the transaction after it's been finalized
  successfully.
  

OPTIONS:
   --funded_psbt value  the base64 encoded PSBT to finalize
   --account value      (optional) the name of the account to finalize the PSBT with
```

Will need to use:
`lncli wallet psbt finalize --funded_psbt "base64_encoded_PSBT"`

Then to publish (do it via LND to avoid any issues):
```
lncli wallet publishtx
NAME:
   lncli wallet publishtx - Attempts to publish the passed transaction to the network.

USAGE:
   lncli wallet publishtx [command options] tx_hex

DESCRIPTION:
   
  Publish a hex-encoded raw transaction to the on-chain network. The 
  wallet will continually attempt to re-broadcast the transaction on start up, until it 
  enters the chain. The label parameter is optional and limited to 500 characters. Note 
  that multi word labels must be contained in quotation marks ("").
  

OPTIONS:
   --label value  (optional) transaction label
```

Will use:
`lncli wallet publishtx "tx_hex"`

rootzoll, 2022-12-13:
OK fpor RaspiBlitz v1.9.0 LND menu will simply show the information of 
```
lncli wallet accounts list --name default
```
and tell people to report their experience bac to this issue. Pushing it to v1.10 milestone to improve on that feedback & for the PSBT part.