raspiblitz/raspiblitz #2193: Remove Specter as WebApp from Menu?

rootzoll, 2021-04-18:
v1.7 introduced the option to connect the specter desktop app with RaspiBlitz .thru the CONNECT menu .. it can make sense to make this the new default to work with RaspiBlitz & Specter ... and to phase out the Specter as WebApp running directly on RaspiBlitz, because its first of all consfusing to have two ways to use Specter with RaspiBlitz.

Also the use of the Webcam for QR code scan between Specter (Web)App & Specter Device makes running HTTPS in Specter WebApp needed AND this means the user sees a "Connection not secure" when calling in local browser and even when runnign in Tor browser this is not a good UX ... the Specter Desktop App can better handle this.

First step rpposed for v1.7.1 is too remove Specter WebApp from RaspiBlitz SERVICES menu. It will be still available for manual install by script until a RaspiBlitz 2.0 .. but that is just for users that had it installed before and have a configuration there they depend on. With v1.9 there should be an export or something.

@k9ert would you agree to this or anything to consider?

openoms, 2021-04-20:
I do prefer running Specter on my node and not exposing the (unencrypted) bitcoin RPC unnecessarily, but curious what others think.

k9ert, 2021-04-21:
One of the major advantages of raspiblitz and specter as well (but not e.g. umbrel) is flexibility on how things can be used. I used to call that "[mechanism, not policy](https://web.archive.org/web/20121223042657/http://blog.koehntopp.de/archives/2069-Mechanism,-not-Policy.html)" (and/or [here](https://www.linuxtopia.org/online_books/programming_books/art_of_unix_programming/ch01s06_3.html) ). It makes things a bit harder because you have an order of magnitude more usecases and scenarios than if you create a crystal clear product which is best use in only one way.

So, similiar than @openoms do i prefer to run my specter directly on the raspiblitz but people have different choices. I'd love if we keep the options here. This is exactly why raspiblitz is much better than umbrel (for many more technical people).

nyxnor, 2021-04-23:
Option is the best thing, connect to the method you find best. Blitz is versatile because of this.

rootzoll, 2021-04-23:
Thanks for the input. So we can leave both options in the SSH GUI for the pros that like having the option. But to reduce confusion for noobs that will start with the v1.8 WebUI - I would just add connecting SpecterApp Desktop.

So if noobs start to dig deeper into the SSH menu they can have the option to run Specter directly on the Blitz.

Is that a compromise that works for all?

openoms, 2021-04-24:
Agree that can't be everything in the WebUI to start with and best is to keep it simple with limited choices.

The strength of the blitz will likely remain to be the advanced usage facilitated with the ssh menu.

Will see how many times we will get the "can I" type of questions and will need to just point to the ssh menu anyway.

k9ert, 2021-05-01:
Yes, agreed! Thanks!

rootzoll, 2022-05-09:
We had Specter also as part of Apps to install planned for v1.8.0 ... but deactivated for now because the self-signed https cert that Specter uses in its RaspiBlitz integration was rejected by chrome browser and so chrome users would not be able to open the App. 

Pushing this to v1.8.1 take a look again. @k9ert was there a special reason for Specter not using the ngnix HTTPs proxy like other apps doing it?

k9ert, 2022-05-10:
> Pushing this to v1.8.1 take a look again. @k9ert was there a special reason for Specter not 
> using the ngnix HTTPs proxy like other apps doing it?

Definitely not. I was simply not aware of it. At least i can't remember thinking about that option. Probably i was busy with other details of the integration so that i simply missed that option.

rootzoll, 2022-05-11:
Specter using its own cert was also mentioned here: #1529 (just to keep track on releated issues)

rootzoll, 2022-11-26:
OK I see no need at the moment for a DesktopApp. Closing.

closed 2022-11-26 by rootzoll