raspiblitz/raspiblitz #2194: Offer HTTP for additional apps (as a bridge solution)

rootzoll, 2021-04-18:
I started to make additional apps available also thru HTTP while showing the HTTPS with the slef-signed fingerprint next to it. See #2157

Because it was very close to v1.7 relase I was not able to make this consistant for all additional WebApps yet - so here are the ones that need still special care for v1.7.1

**Specter** --> Its in the discussion to remove from WebApp anyway (move to connect as DesktopApp) because HTTPS is a MUST otherwise WebCam access is not possible and we will end up with a self-signed cert waring anyway in v1.8 even when running within Tor browser. See #2193

**Lightning Terminal** --> Is using its own HTTPS derver by default. We need to edit it to serve HTTP by default and then integrate it into the NGINX setup with SSL like the other apps. To switch to HTTPS see https://github.com/lightninglabs/lightning-terminal/issues/162 ... we need todo this anyway to allow serving over Tor without any self-signed warnings for v1.8

**BTC-Pay-Server** --> Not sure why yet .. had problems to quickly offer HTTP access to the outside.

supertestnet, 2021-09-16:
Here is a scenario that just happened:

- User buys a raspiblitz
- User sets up lnbits
- User creates paycode
- User scans paycode
- Wallet gives them error about https://tordomain.onion
- User complains to supertestnet

I asked the lnurl developers about it and they said the lnurl spec doesn't allow https://tordomain.onion and that that is intentional. They indicated that raspiblitzes shouldn't serve lnurls over https. Apparently lnbits can be configured to serve over http, is this something you can fix?

rootzoll, 2021-09-16:
LNbits needs https to have access as to the Webcam for the Fallback case for LNURL vouchers. I dont have a test setup here at the moment but as far as I remember the LNURLs worked over https as long as its a valid certificate like with LetsEncrypt. And to set this up thats why there are the IP2Tor & LetsEncryt features in RaspiBlitz ... see: https://www.youtube.com/watch?v=0Bt3tHULAnw

If I get you right the problem here is with the Tor-Domain? To have LNURL working over Tor a wallet needs to support Tor ... thats not possible yet with most of the wallets that support LNURL as far as I know. It would be great if someone could research this a bit deeper ... like making a matrix what Lightning-Wallets can already support LNURL over Tor out-of-the-box. Then we could see what special setups are needed to deliver towards that.

supertestnet, 2021-09-16:
Lnurls don't work when served over an onion address that has an https cert. It violates the lnurl spec to use an https cert with a tor domain. Any wallet that follows the lnurl specification will regard such lnurls as invalid.

The problem is not with the tor domain by itself, it is with a tor domain that has an https cert. I used a wallet that supports tor (Simple Bitcoin Wallet) while connected to tor via Orbot. The wallet gave me an error about the https cert because the use of an https cert with a tor domain violates the lnurl specification. I spoke with the lnurl devs in their telegram channel and they said the lack of support for https tor domains is intentional and any service that serves lnurls over https tor domains is acting in violation of the lnurl standard. That's when I added to this issue.

frennkie, 2021-09-17:
@supertestnet What exactly is the problem? RaspiBlitz hosts LNbits over Tor both over HTTP and over HTTPS. If LNURL requires either HTTPS [x]or TOR+HTTP... then this should work with the default RaspiBlitz setup... or where does this break?