raspiblitz/raspiblitz #2195: Rapsiblitz webadmin and RTL GUI fails

silvester235, 2021-04-18:
I finished to set up the raspibliz today.

However I use it without LCD and therefore would like to use the webadmin.

Loged in via ssh and the "RaspiBlitz Status Screen" shows:
Web admin -> http://10.0.1.9:3000

If I enter http://10.0.1.9:3000 in the Chrome Browser it brings me to RTL GUI.

--------
RTL:
"Open in your local web browser & accept self-signed cert: https://10.0.1.9:3001"

Error in the Chrome browser "This site can’t be reached- ERR_CONNECTION_TIMED_OUT"

So I can't log in to the Web admin at all. And RTL login works with at a different port but only as http not as https.

I am confused now.
Is there a way to fix this?

[image]
[image]

silvester235, 2021-04-18:
Here are the RASPIBLITZ DEBUG LOGS 

https://termbin.com/gktg

rootzoll, 2021-04-18:
There is no WebAdmin for RaspiBlitz itself yet .. just a stub for testing. A full WebUI is targeted for v1.8 now.

The RTL timing out on https is a but strange - when its still reachable (and i dint see an error in logs for RTL) it might be a problem with nginx. 

Best way would be wait some more days for the v1.7 (where RTL has an upadte anyway) and then report back if that fixed it for you.

silvester235, 2021-04-18:
Thanks for the update.

nginx is the problem indeed. 

My local mirror has an error in the URL structure. Err:1 http://mirror1.ku.ac.th/raspbian/raspbian instead of http://mirror1.ku.ac.th/raspbian resulting in 404

Is there a way to report the wrong mirror URL structure?

nyxnor, 2021-04-19:
For nginx
```
sudo nginx -t
sudo systemctl status nginx
sudo systemctl restart nginx
```

For the mirror (delete wrong info)
```
sudo nano /etc/apt/sources.list
deb http://mirror1.ku.ac.th/raspbian/raspbian/ main contrib rpi non-free
```

nyxnor, 2021-05-03:
pls close this if it worked.

closed 2021-05-03 by silvester235