raspiblitz/raspiblitz #2198: Connecting Mobile Apps

rootzoll, 2021-04-18:
The mobile Apps are having problem connecting to the RaspiBlitz.

ZAP Android --> Its seems its not accepting the LND with the new TLS settings. Need to do further tests.
Zeus --> is not connecting to the Tor services ... says its offline - can be someting with my iOS test device. Need to do further tests.

rootzoll, 2021-04-18:
@openoms could the zeus app not connecting anymore be a side effect of #1604 .. will check in the Tor addresses

openoms, 2021-04-18:
unlikely.
The Hideen Service for 8080 is managed by tor@default and the default torrc, no changes there.

The34Goon, 2021-04-19:
I just went through the installation process again and had no trouble connecting Zeus-IOS v0.5.1 to 1.7RC2 over Tor.

Bonus: Nice to finally see the complete QR in fullscreen-terminal.

rootzoll, 2021-04-19:
OK I got Zeus running over Tor - all cool there.

But Zap-Android has the problem that it requires the IP in the TLS certificate and seems not to play nice with the new LND settings. @openoms I think I will remove also Zap-Android for v1.7 .. when someone likes it back we might need to open a issue on the Zap-Android repo on it.

closed 2021-04-19 by rootzoll