raspiblitz/raspiblitz #2199: Add Telegraf WebUI

rootzoll, 2021-04-18:
Experimental telegraf background service was added with #1369 ... see if its possible to make it a permanent install (add to provison for updste/recover) & see if we can add a WebUI integration.

PatrickScheich, 2021-04-19:
**telegraf** is just a metric collecting agent. Normally it pumps the measurements into a backend infratructure.
Often a timeseries database like **influxdb**, on top of that DB sits a "nice graphic frontend" like **grafana**.

I'm currently not aware of any lightweight WebGUI for the telegraf output.

In my environment I use a seperate Raspi4 as InfluxDB and grafana host.
So it could be run on a respberry pi but I would not recomend to run it on the raspiblitz.

holgern, 2021-04-20:
It seems to me that questdb could be a lightweight alternative?
https://questdb.io/docs/third-party-tools/telegraf/

holgern, 2021-04-20:
I think when using a retention-policies for influxdb which deletes old data automatically, the memory usage should be fine.

https://docs.influxdata.com/influxdb/v1.7/query_language/database_management/#create-retention-policies-with-create-retention-policy

closed 2024-08-06 by rootzoll