raspiblitz/raspiblitz #2201: LCD Screen not working

mantiev, 2021-04-18:
I'm a noob to raspiblitz and can't seem to get my LCD screen to work.  I have installed the drivers and done a few other things I have found in closed issues to no avail.  I even returned the LCD I had to make sure it wasn't bad.  I'm getting the same bluish white screen but nothing displays and I have raspiblitz completed installed an synched.  Any help would be greatly appreciated.

nyxnor, 2021-04-19:
Always provide important info
Is it the recommended screen? What is the brand and model of the screen? What is the link for the drivers to be installed? Is it the recommended by the manufacturer? What is the link you got the screen from? Did you manage to get it working without a scree?

mantiev, 2021-04-19:
Osoyoo 3.5 Touch Screen LCD - https://github.com/kedei/LCD_driver is where I got the drivers from as advised by the manufacturer.  

https://www.amazon.com/gp/product/B07TPRX8M2/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I have managed to get the raspblitz setup completely without the screen.  I even checked off Touchscreen in the settings but no display and nothing happens when I try to calibrate the screen.

nyxnor, 2021-04-19:
It is not the recommended screen. You are indeed using the script recommended by the seller. Just make sure you've done it right 
https://osoyoo.com/2019/06/04/osoyoo-128m-spi-lcd-touch-screen-for-raspberry-pi-installation-guide/

mantiev, 2021-04-19:
Thanks for the assistance - went back through and same issue.  I will order the recommended screen and try again.

closed 2021-04-19 by mantiev

nyxnor, 2021-04-19:
@mantiev Make sure to send back to the buyer your screen so you can have more fiat to buy sats. Also wait a bit before ordering a screen, Blitz with raspios_arm64 is going through some things to calibrate the screen, you should wait a bit to see if this works first.

nyxnor, 2021-04-22:
@mantiev Update, new information that the touchscreen is working now.