raspiblitz/raspiblitz #2209: Improve TorByDefault

rootzoll, 2021-04-19:
Most basics for TorByDefault #2056 got covered with the realease of v1.7 .. for v1.7.1 there are some special cases to look into again and to improve: 

- [ ] https://github.com/rootzoll/raspiblitz/issues/1341
- [ ] https://github.com/rootzoll/raspiblitz/issues/2013
- [ ] https://github.com/rootzoll/raspiblitz/issues/2054

maxgoldberg, 2021-05-01:
Would also like to see an option to enable ipv4/ipv6 in addition to tor instead of it just being a binary tor-only option. I'd like the option to have my node be as available as possible.

rootzoll, 2021-05-01:
@maxgoldberg basically that makes sense to have that option, but because of time i dont think we will be focus on this already for the v1.7.1. and the v1.8 also has mainly focus on UI/UX - so I created this as an issue for v1.8.1 - see: #2259

But of course you can already manually config your LND under SYSTEM > LNDCONF .. please share your experiences in adding more networks manually in this new issue: #2259

janis-mueller, 2021-05-27:
@rootzoll and would be great to give the option to connect to a mobile wallet using IPv6  (public IP) or TOR by choice if both options are activated

rootzoll, 2021-12-11:
On the tor branch `2669tor` where I merged now the tor changes with the changes on dev I was able to build a new sd card and setup a fresh LND node with it - but LND in the end is not able to start (wallet unlock, shutdown, repeat).

The error in the logs is: `unable to initialize tor controller: unable to connect to Tor server: dial tcp 127.0.0.1:9051: connect: connection refused`

While `sudo systemctl status tor@default` is showing a long running and 100% bootstraped tor service.

@nyxnor @openoms any quick hint what to check?

openoms, 2021-12-11:
check the torrc
the tor cookie needa to be group readable
and
add the `bitcoin` user to the `debian-tor` group

rootzoll, 2021-12-11:
Thanks was a good hint ... I had a typor that didnt triggered the `tor enable` part and so missed to edit the torrc.

rootzoll, 2021-12-14:
OK Tor seems now working for LND in branch `2388redis-tor-dev-merge`

@nyxnor but looking into the `/etc/tor/torrc` after enable I see the line that looks a bit like a broken path, is that correct that way? 
```
%include /mnt/hdd/app-data/tor//torrc.d
```

nyxnor, 2021-12-14:
it is wrong, it should have no trailing '/ or `tor_conf_dir`
https://github.com/rootzoll/raspiblitz/blob/42593a94f06180992b8253ba639f6a225b63aa9f/home.admin/config.scripts/tor.install.sh#L26

correct:
`tor_conf_dir="${hdd_path}/app-data/tor"`

rootzoll, 2021-12-14:
OK looks better now `%include /mnt/hdd/app-data/tor/torrc.d`