raspiblitz/raspiblitz #2210: Error formating HDD

ronledfordjr, 2021-04-20:
[image]
[image]

Error formatting HDD with new BTRFS.

This is my third node and I have not had any issues to this point.

[image]

nyxnor, 2021-04-20:
run `lsblk` to see available drives.

ronledfordjr, 2021-04-22:
[image]

nyxnor, 2021-04-22:
You thumb drive or second disk was mounted first as sda, and your SSD was mounted second as sdb.
Try again just with the SSD and the micro SD card attached.

rootzoll, 2021-04-23:
Yes please make sure to not have a thumbdrive connected while doing a BTRFS format for now.
I will mark this issue for v1.8.1 when we will start to check back on BTRFS again.

21-btcclub, 2021-05-01:
I have a similar error
/home/admin/XXdebugLogs.sh | nc termbin.com 9999
****
and when i start without the usb stick, raspiblitz will Format in EXT4 layout (old)
How can i use BTRFS format without a thumbdrive connected?

closed 2021-05-01 by ronledfordjr