raspiblitz/raspiblitz #2215: no HDD - usb-storage module loading to late

rootzoll, 2021-04-21:
There was one report from a v1.7RC3 tester that the RaspiBlitz on startup was not detecting the SSD and it maybe because the usb-module is loading too late:

```
[    1.373207] usbcore: registered new interface driver r8152
[    1.373280] usbcore: registered new interface driver lan78xx
[    1.373333] usbcore: registered new interface driver smsc95xx
[    1.374940] pci 0000:00:00.0: enabling device (0000 -> 0002)
[    1.374972] xhci_hcd 0000:01:00.0: enabling device (0000 -> 0002)
[    1.375066] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    1.375105] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    1.375838] xhci_hcd 0000:01:00.0: hcc params 0x002841eb hci version 0x100 quirks 0x0000030000000890
[    1.376776] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
[    1.376791] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.376804] usb usb1: Product: xHCI Host Controller
[    1.376815] usb usb1: Manufacturer: Linux 5.10.17-v8+ xhci-hcd
[    1.376827] usb usb1: SerialNumber: 0000:01:00.0
[    1.377497] hub 1-0:1.0: USB hub found
[    1.377591] hub 1-0:1.0: 1 port detected
[    1.378217] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    1.378238] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[    1.378259] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
[    1.378676] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.10
[    1.378690] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.378702] usb usb2: Product: xHCI Host Controller
[    1.378714] usb usb2: Manufacturer: Linux 5.10.17-v8+ xhci-hcd
[    1.378725] usb usb2: SerialNumber: 0000:01:00.0
[    1.379338] hub 2-0:1.0: USB hub found
[    1.379412] hub 2-0:1.0: 4 ports detected
[    1.380991] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    1.381234] dwc_otg: FIQ enabled
[    1.381245] dwc_otg: NAK holdoff enabled
[    1.381255] dwc_otg: FIQ split-transaction FSM enabled
[    1.381268] Module dwc_common_port init
[    1.381771] usbcore: registered new interface driver uas
[    1.381870] usbcore: registered new interface driver usb-storage
```

rootzoll, 2021-04-21:
@Emzy or @openoms have you an idea how we can improve on this situation.

If this just keeps a single report I think its OK to move this to v1.7.1 - even if this would make a change to the sd card build script its not expected to change any configuration other scripts building on.

rootzoll, 2021-04-21:
@holgern can you check on your sd card if there is a `usbquirks` entry in your `/boot/cmdline.txt`?

openoms, 2021-04-21:
I am testing the RC3 with the X825 board and Sandisk Ultra 1TB SSD. No issues so far.

holgern, 2021-04-21:
There is no usbquirks entry in my /boot/cmdline.txt

I have the following adapter:
ID 174c:55aa ASMedia Technology Inc.

I tried the following:
```
$ echo options usb-storage quirks=174c:55aa:u | sudo tee /etc/modprobe.d/blacklist_uas.conf
$ sudo update-initramfs -u
$ sudo reboot
```
and after it did not worked I added
```
usb-storage.quirks=174c:55aa:u
```
to /boot/cmdline.txt
But it did not help, my USB drive is not detected on boot.

Putting my ssd to a USB2 port did also not help.
I think it is caused by the ASMedia chip, I will switch my ssd - usb3 adapter to a different vendor.

holgern, 2021-04-21:
I tried also to reflash 1.7.0 RC3 and edited cmdline.txt, but it did not helped
There is no the following dmesg:
```
[    2.155407] systemd[1]: Detected architecture arm64.
[    2.177218] systemd[1]: Set hostname to <raspberrypi>.
[    2.680653] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    2.701703] usb 2-1: New USB device found, idVendor=174c, idProduct=55aa, bcdDevice= 1.00
[    2.701725] usb 2-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[    2.701738] usb 2-1: Product: ASM225
[    2.701750] usb 2-1: Manufacturer: ASMT
[    2.701762] usb 2-1: SerialNumber: 000000000001
[    2.704807] usb-storage 2-1:1.0: USB Mass Storage device detected
[    2.705353] usb-storage 2-1:1.0: Quirks match for vid 174c pid 55aa: c00000
[    2.705547] scsi host0: usb-storage 2-1:1.0
```
but no drive is detected

holgern, 2021-04-21:
After starting from scratch and adding
```
usb-storage.quirks=174c:55aa:u
```
to  /boot/cmdline.txt before starting it the first time, everything works now.

rootzoll, 2021-04-23:
At the moment the RaspiBlitz tries to get the HDD adapter ID on first boot and adds it to the `usb-storage.quirks` if its not an adapter from our shopping lists to avoid UASP problems.

You can see that part if the code here:
https://github.com/rootzoll/raspiblitz/blob/40e554c5e721087115a150e1d09f8e71c57197bf/home.admin/config.scripts/blitz.datadrive.sh#L348

@holgern strange thing is the `ASMedia Technology Inc` with the id `174c:55aa` is exactly the adapter from our shopping list that I am testing & developing on and I never had that issue. 

I am not exactly shure why this is happening.

holgern, 2021-04-23:
Strange,
I will test the next release for 1.7.0 without modified cmdline.txt, and report again.
Somehow, my HDD adapter was not correctly identified on first boot. With the modified cmdline.txt, my HDD adapter was detected and was visible in the first lines of dmesg.

Maybe we should use this issue to collect similar cases.

holgern, 2021-04-25:
It worked for me out of the box for the same HDD adapter with the final release .1.7.0

rootzoll, 2021-04-27:
Hmm .. good to hear that it worked with the v1.7.0 ... not sure why. Keeping this issue open on v1.7.1 just in case someone has a similar problem.

rootzoll, 2021-09-20:
No one reported anymore ... closing issue.

closed 2021-09-20 by rootzoll