raspiblitz/raspiblitz #2221: RPI 4b+ with HDMI type monitor has small text

jeffersonn-1, 2021-04-22:
I had to do an edit to config.txt and set the HDMI type to basically 640x480 so as to get the screen to be readable ... This is on 1.7RC3

rootzoll, 2021-04-23:
Thanks for testing .. can you share your edit? Could make sense to add this when users are switching to HDMI automatically for v1.7.1

jeffersonn-1, 2021-04-23:
in config.txt 

hdmi_group=1
hdmi_mode=3 
# This sets it to essentially 640x480 mode ... It's not a perfect fix but hella better than 1920x1080 and tiny.
gpu_mem=128
#I also boosted vidram to 128 - the refresh rate works better there for some reason at 2mb it would freeze  -- it is possible that 64mb would work. All I know is at 2mb it would work for ~15 mins and then stop. 

Complete reference here: https://www.raspberrypi.org/documentation/configuration/config-txt/video.md
Honestly - if there was a way to more easily edit config.txt ... that would be awesome. I am not sure how that whole thing works (how the fonts / display of it works in relation to the HDMI output, etc.) ... The unit that it is attached to is a non interactive (non touch) panel that can do 1900x1080 max resolution on HDMI. 

[https://imgur.com/a/9iT7AeT](Picture of the unit
[image]
)

rootzoll, 2021-04-23:
Thanks @jeffersonn-1 added this to the v1.7 release. Tests looking good - closing issue for release.

closed 2021-04-23 by rootzoll