raspiblitz/raspiblitz #2222: Setting/Option to turn off RPi system LEDs

Vanmar1, 2021-04-23:
Not really an issue, more than a quality of life.

How tough would it be to add a setting to turn off the system LEDs to the 1.7 update?
No rush, was just something I had to do some searching for to turn it off and will probably have to re-do it for the 1.7 update.  Maybe.

Thanks though!

rootzoll, 2021-04-23:
If you can find a link that shows how to-do that or have code/edits to share that would be a nice option for the future.

NilByte, 2021-04-23:
There is a thread here referring to how to turn leds off.

https://github.com/rootzoll/raspiblitz/issues/1564

nyxnor, 2021-04-23:
@Vanmar1 I wrote this some time ago on telegram but I dont have lcd to test

From here https://raspberrypi.stackexchange.com/questions/84401/turn-pi-screen-on-and-off-programmatically
Command line:

OFF:
`sudo -E sh -c 'echo 1 > /sys/class/backlight/rpi_backlight/bl_power'`
or
`echo 1 > /sys/class/backlight/rpi_backlight/bl_power`

ON:
`sudo -E sh -c 'echo 0 > /sys/class/backlight/rpi_backlight/bl_power'`
or
`echo 0 > /sys/class/backlight/rpi_backlight/bl_power`

TIP: I define the above as backlight_off and backlight_on in ~/.bash_aliases. Be sure to save that file elsewhere, since you'll lose it if you re-image.

Never saw someone here mention this, so try with caution. Check if it is the same file, also notice this is for RPI3B, could be different on the 4.