raspiblitz/raspiblitz #2228: Raspberry Pi CM4 and PCIe NVMe SSD

normandmickey, 2021-04-25:
I finally received a Raspberry Pi 4 Compute Module (CM4) and carrier board that I ordered back in November.   The CM4 comes in many flavors but the top of the line board has 4GB of RAM and a 32 GB EMMC ($65).  The official carrier board costs $45 and has a PCIe 1x slot so you can add a 1TB NVMe SSD.   There are also several other promising carrier boards being developed.  Here's some links to one example.  
https://github.com/mfolejewski/MirkoPC
https://youtu.be/DZMF1n75Km0

A recent update to the Raspberry Pi OS enables NVMe support by default.  It may be worth looking into adding NVMe support to Raspiblitz.  When enabled the first NVME drive shows up as /dev/nvme0n1p1.

rootzoll, 2021-04-27:
Thanks for the report .. I think after v1.8 it can make sense to see what more professional hardware options we can combine for a more pro setup. Detecting and using an NVM could be intergrated earlier, if someone would like to to add this to the `blitz.datadrive.sh` script.

BitcoinLabrador, 2021-07-08:
I got it to work with some instructions from openoms. Here's what I did:
First, look up the disk name (probably /dev/nvme0n1): `lsblk`
Then create a new partition table on the disk, use default values: `sudo fdisk /dev/nvme0n1` (don't forget 'w' when you're done to write the changes to disk)
Format a new partition & give it the "BLOCKCHAIN" label: `sudo mkfs.ext4 /dev/nvme0n1p1 -L BLOCKCHAIN`
Find the PARTUUID of your new partition and copy it: `blkid`
Make sure it mounts during boot: `sudo nano /etc/fstab` append `partuuid=abcd-1234 /mnt/hdd ext4 default 0 0` save `ctrl-x y enter`
Reboot & SSH back into your node: `sudo reboot`

normandmickey, 2021-07-18:
In the "blitz.datadrive.sh" script they are running "lsblk" and looking for device names that start with "sd".  The script doesn't find the NVMe disk because it gets assigned a name that start with "nvm" (mine is "nvme0n1").

Here's the command in the script that looks for the disk name (sda1, sdb1, etc.)
lsblk -o NAME,SIZE -b | grep -P "[s|v]d[a-z][0-9]?" > .lsblk.tmp

I only have one CM4, carrier board and NVMe disk and they're being used for something else so I can't test this but I think the command should be changed as follows. 
lsblk -o NAME,SIZE -b | grep -P "[s|vn][dv][a-z][0-9]?" > .lsblk.tmp

rootzoll, 2021-07-20:
I added the proposed changed line by @normandmickey to the script.

I dont have the needed hardware here to test. The change will be part of v1.7.1 RC1 (a very eraly test realease coming soon) - it would be great if someone could verify if that solved the problem.

BitcoinLabrador, 2021-07-20:
I already changed the line in my file, and it didn't solve it unfortunately. Still working through the code to see how I can get this to work.

rootzoll, 2021-09-07:
The change above will be part of v1.7.1 - but I push this issue up to later because I would need the hardware first to test this.

rootzoll, 2022-12-18:
OK #340 is merged for v1.9.0 release - it should also solve the NVMe pert for the PiCM 4 - but I have no board here to test.

_At least for me at first glance it should solve it. The script now looks for nvme in front of the device name and if it's there it parses the data and partitions using nvme formats instead of sata. It works on the nuc I have, and it also works on my pi, but the pi doesn't have an nvme drive so I technically can't test it with a pi AND nvme at the same time._

BitcoinLabrador, 2022-12-19:
I can't test it for at least a few months, sorry. I'm traveling and my node is at home.

closed 2023-11-14 by rootzoll