raspiblitz/raspiblitz #2231: notifyme no longer works

frennkie, 2021-04-25:
Just noticed that the sstmp package is no longer included in the updated base image.

```
admin@raspberrypi:~ $ sudo apt install ssmtp
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package ssmtp is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ssmtp' has no installation candidate
```

Need to find an simple alternative.

dennisreimann, 2021-04-26:
I also came across this while running `/home/admin/config.scripts/blitz.notify.sh on` after the v1.7.0 upgrade.

frennkie, 2021-08-03:
Did you see #2232 ...?

vv01f, 2021-08-03:
^^ no, I searched for msmtp and didnt find mstmp ;P. well, close my pr for now and wait for this to merge. then I can re-apply sendxmpp on that later.

rootzoll, 2021-09-14:
PR merged to be included in 1.7.1RC2 ... set to final testing.

rootzoll, 2021-10-09:
no bug reported on rc2 ... so closing for final release.

PR #2232 (2021-09-14): switch to mstmp
fix #2231
closed 2021-10-09 by rootzoll