raspiblitz/raspiblitz #2239: v1.7.0 Litecoin installer stuck at LND unlock

Himbeergeld, 2021-04-27:
Updated my Bitcoin and Litecoin nodes to v1.7.0, both RasPi 3B+. Bitcoin node update worked seamless but Litecoin node installer always gets stuck at LND unlocking.
After entering the Litecoin Lightning wallet password, it gets stuck at
"calling: lncli unlock"
Already rebooted twice, but always gets stuck at this line. Does anyone else have the same issue with Litecoin?

rootzoll, 2021-04-27:
Marked it for v1.7.1 to be re-tested and to target for a possible fix.

@Himbeergeld have you tried to unlock from terminal with `lncli` diretcly?

Himbeergeld, 2021-04-27:
At previous attempts I connected via SSH >during< the LND unlock request, then after entering the password it got stuck. 
Now tried again but connected via SSH >before< the LND unlock request. Now it worked fine. Closing this issue.

closed 2021-04-27 by Himbeergeld