raspiblitz/raspiblitz #2240: v1.7.0 LndHub Url shows 404

klnit, 2021-04-27:
I am using the LndHub feature of LNbits to connect the Bluewallet App. Since Release 1.7.0 this the url shows "404 not found". Maybe it is a LNbits issue? It happens with duckdns + Ip2Tor as well as locally.
Can you reproduce it?
E.g. the admin url is lndhub://admin:1b96.............e5cc66aff@https://example.duckdns.org:30803/lndhub/ext/
The part https://example.duckdns.org:30803/lndhub/ext/ shows 404

rootzoll, 2021-04-27:
If your LNbits otherwise is working behind that `https://example.duckdns.org:30803` IP2Tor forward, then it looks like an LNbits issue. Best would be to hop over to their GitHub and report back: https://github.com/lnbits/lnbits

klnit, 2021-05-01:
LNBits guys say there should be no resonse at this url. Bluewallet is adding paths...
I was probably mislead.. assume it is #2247 (because I have the same error message "Not able to get HTTPS response")

frennkie, 2021-05-01:
@klnit as I mentioned in the other issue.. is this actually an "error".. or just a warning..?!

klnit, 2021-05-03:
Hi frennkie,

probably just a warning. I get the message when opening the LNBits menu item as in #2247.

Maybe a little more content... I first noticed that my duckdns url was not accessible in Firefox anymore. And access from my Bluewallet App to the LNBits lndhub extension was not working. So I assumed some problem in the extension or IP2Tor/duckdns/Letsencrypt area.
But I guess Firefox just does not accept self signed certificates anymore because Chrome and Edge are working. Something similar could be on the Bluewallet side.
Luckily Bluewallet made a major update yesterday and they integrated Tor. So to connect to the LNBits lndhub wallet we can now use the onion address :-)

So this is working for me and I close this issue
Regards
PS: @bluewalletio and @getumbrel are making a lot of noise on twitter with this update. Time to do some advertising guys

closed 2021-05-03 by klnit