raspiblitz/raspiblitz #2241: Donation page error

Nerfzooka, 2021-04-27:
I followed the donation [link](https://tallyco.in/s/r5lx23/) to apply some appreciative sats, touched the lightning button, but got an error.  

[image]

rootzoll, 2021-04-27:
Thanks for the report .. its a reminder from the universe to get the `Integrate Tallycoin Connect` feature for RaspiBlitz done see #1450

nyxnor, 2021-05-03:
Someone donated on-chain yesterday, off-chain still not working.

rootzoll, 2021-05-18:
Because for legal/tax reasons it makes sense to handle donations like mentionen here #2292 anyway.

nyxnor, 2021-05-18:
Just one thing, you are going to use their node, not yours, and "IRS".
DM with Ben Prince from OpenSats (abitcoinperson on twitter, did not find his username here), this is his responses, I cut my questions:
>Our model, as of now, would be running our own btcpay instance. Each project has its own “store” and any donations to that store are tracked individually. 100% of the Sats would be passed on to the project with no fees. So generally same outcome with us in between (for 501c3 purposes) plus we add in fiat donations which will be converted and distributed in Sats
Unfortunately the irs has so many * rules
One of which prevents us from “directly” passing through donations. We need to make sure that we get projects to ensure they are helping fulfill our mission, which is very broadly written
Already kinda playing in a grey space here
That being said, I’m definitely gonna ask about the idea of ppl hooking in directly to the platform with their nodes
I think we might let them directly open channels and do payouts that way
At least that was my v1 idea.
Since we’re only paying out Sats, it’d make sense to use L2 when possible

Enphasize: This was what he described as a initial proposal, not definitive.

For legal/tax reasons, donations on a hiddenservice receiving with a nym :)
If being donated bitcoin, don't handle it to the mafia, I would be a indirect contributor to this atrocity by donating with this rules. Blitz and Bitcoin are not obliged to comply to constitutions, it just runs the code.
Please reconsider this.

rootzoll, 2021-05-19:
Hmm .. OK sounds it adds more problems to the mix then it solves.