raspiblitz/raspiblitz #2242: SD-CARD: Non-Interactive Setup Process

rootzoll, 2021-04-27:
In preperation for v1.8 the v1.7.1 will contain a refactoring of a non-interactive setup process #1126 that does not require any changes to the sd card. Check for v1.8 release if any changes needed in the sd card build to optimize that further.

nyxnor, 2021-04-28:
Is there a discussion for me to read about the build_sdcard.sh? Because I think many things dont need to be there (I will last the lines later), for example, https://github.com/rootzoll/raspiblitz/blob/2a89fae133bcf4e0a4b66557750ebdf0902f7f39/build_sdcard.sh#L933
What if the user wants c-lightning and not lnd? Yes, it is just the gpg, but still. I think this should be a blitz.lnd.sh (unifying it with lnd.update.sh as the option inside).

rootzoll, 2021-04-28:
In the beginning it was the idea that the sd card build will not call other scripts - so thats its easier to code check. But the build script got so advanced now that it makes sense to push some parts into scripts just even to prevent code duplication - especially now that we allow more config options on sd card build.

So next release where this is possible is the v1.8.

nyxnor, 2021-04-29:
Ok, maybe even Tor and all extras that does not interfere with the system should be removed from the build.
After flashing the sd card, the first step to activate the other process should be through ssh or webui and toggling it to `on`. Also, the parameters of the build script should also be moved after the user can access the blitz interface, because if someone flashes your image, they are set to your parameters, and the way it is done, he wont be optimized unless he builds the image, but with a reduced image config, all parameters being set after. This will reduce drastically the dependency on the build and just the scripts called after. It is good that the build dont call any other scripts. Using this options, the updates will just get the new scripts, set permissions, and some things more.

Oh, but what about building from your forked repo? https://github.com/rootzoll/raspiblitz/blob/2a89fae133bcf4e0a4b66557750ebdf0902f7f39/build_sdcard.sh#L53
It still will be available, the minimal blitz will give access to the webui and then the user can choose there, with an option of using the default values of the main project repo and other configs.

rootzoll, 2021-04-29:
I nice way to add network configurations to a sd card build is also to write a file to the sd card when connected to your laptop .. like its done for the WIFI config.

But basically if the RaspiBlitz can detect itself that it needs additional Tor config - it could be done also on the WebUI start page that you call with your normal browser locally. There it tells you to install Tor Browser and and yould also tell you that more Tor config is needed, etc.

rootzoll, 2022-05-09:
I think we had the major refactors of the build script already fixed. @nyxnor I would close this issue but if you see ideas in here that are not reflected by other open issue please create a new one.

closed 2022-05-09 by rootzoll