raspiblitz/raspiblitz #2243: Recommended screen no longer available

pandascrape, 2021-04-28:
The screen in the recommended kit list has not been available on Amazon for some months now.

Unfortunately there does not appear to be another compatible screen that is as cheap, however this screen and case appears to be available in both US and Europe: https://www.amazon.com/Raspberry-Touchscreen-320x480-Monitor-Display/dp/B07WQW6H9S

Please could you confirm that this should be compatible, and perhaps update the recommended kit list on the GitHub repo?

And thanks for the great work. You are bitcoin heroes :)

rootzoll, 2021-04-28:
@Animalmother9000 can you please create more international Links to alternative LCD models that we know that work for the shopping list? You can just post them here and I can add them top the GitHub Repo.

NilByte, 2021-04-29:
I'm using this one and have not had any problems with it.

https://www.amazon.de/gp/product/B06X191RX7/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1

rootzoll, 2021-04-29:
OK the PR #2253 gives now some alternative models to search for. See: https://github.com/rootzoll/raspiblitz#amazon-shopping-list-buy-parts--build-it-yourself

Closing issue.

closed 2021-04-29 by rootzoll