raspiblitz/raspiblitz #2244: Problems with update to 1.7 (and power-cut/blackout on 1.6 before that)

KainerW, 2021-04-28:
Before updating: Still on 1.6 I had a power-cut/blackout. Raspiblitz showed error message... wasn't able to run anything or fix that. Error message on screen: "Give root password for maintenance (or press Control-D to continuine)" CTRL D did nothing and giving my passowrd did not help.

So I thought: ok, electricity blackout bricked the sd card with 1.6 on it, I want to update to 1.7 anyway...

First try: installed/flashed 1.7 on a brand new sd card (microsdhc class 10). Put that into the raspiPi. 

Result after more than 4-5 hours: Node running. Lightning channels visible. Specter Wallet running. But: RTL, Thunderhub and Sphinx chat all showed error in normal-browser & tor-browser. 

So I did it again, flashed another brand new sd card with 1.7 (same model from same supplier) -> result: now Thunderhub and RTL running ... but Specter-wallet, Sphinxchat and Mempool not running, showing errors in browser and Tor-browser.

What helped now to get Specter wallet running was deleting and installing it again on the raspiblitz (only specter wallet, not the whole raspiblitz software).

I can not do the same for sphinx chat as I lose all my tribes/chats then. The problem with sphinx chat seems to be the dns https service. My subscriptions are showing as active/running in raspiblitz but I get in the browser "Cannot get /" when trying to reach my domain (setup before via ddns/dynu and working fine under 1.6).

I tried to restart the node a few times. Did not help. Main problem for me right now: sphinx chat relay not working. Showing: Problem detected:  "Not able to get HTTPS response." Cheining domain in Tor only shows: "INDEX".

Maybe the power-cut/blackout destroid a part on the SSD-Harddisk which makes all the update to 1.7 buggy? What might be interesting: As raspiblitz 1.7 tried installing mempool and specter it took more than 1.5 - 2 hours for each software in the installation provcess. And I know it normally takes only 15-20 minutes. 

Any idea how I can solve it to get sphinx chat relay up and running without losing my chats and tribes?

rootzoll, 2021-04-28:
Just to try to rule out the power-cut side effects: If you are willing to invest into a new/second SSD you can use `REPAIR > MIGRATION`, download the migration file that contains all app-data (also your sphinx data, etc) and then shutdown. 

Now you make a fresh v1.7 sd card (if you can afford also use a new second one) & connect the new SSD and setup a fresh RaspiBlitz BUT on first dialog instead of choosing BITCOIN you choose `MIGRATION` and upload your migration-file to the fresh hardware. It will need to sync the blockchain again after that - but then you are runnign on a fresh SSD.

KainerW, 2021-04-28:
OK. That would take me another investemant of money and time. I will report back in ~14 days. 
Buying a USV now too.

nyxnor, 2021-05-17:
@KainerW 
> I will report back in ~14 days.

Hi! If this was solved, please close.

nyxnor, 2021-05-17:
If this was solved, please close.

rootzoll, 2021-05-18:
Closing. Please reopen if still an issue.

closed 2021-05-18 by rootzoll