raspiblitz/raspiblitz #2246: LCD35B and UPS issues for version 1.7.0 (hand fixed)

intorid, 2021-04-28:
After upgrading from 1.6.0 to 1.7.0, Raspberry Pi 4 the method for recovery a recorded image from a flash drive.

1. issues: LCD35B rev.1 (old version)

Previously post https://github.com/rootzoll/raspiblitz/issues/493 worked well on Raspiblitz 1.6.0  there was a similar problem with my screen, the model 3.5inch is original: Waveshares 3.5inch RPi LCD (B) Rev.1 (old version). The installation was done according to the [wiki](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)):  
`git clone https://github.com/waveshare/LCD-show.git`
`cd LCD-show/`
`#if old version, use this command:`
`chmod +x LCD35B-show`
 `./LCD35B-show`
But with the upgrade to version 1.7.0, the RPi4 core version was updated to 5.10 and the problem (inverted color) came back again, now the official wikki solution doesn't work! Waveshares 3.5inch RPi LCD (B) Rev.1 (old version) display is discontinued.
The next step, found a working solution. Took use of third party git clone repository @swkim01 fork with dtoverlay option enabled: https://github.com/swkim01/waveshare-dtoverlays.git
The **waveshare35b** driver for [Waveshares 3.5inch RPi LCD (B) Rev.1 (old version)](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)) was used, my setup:
`sudo git clone repository solution https://github.com/swkim01/waveshare-dtoverlays.git`
`sudo nano /boot/config.txt`
Enabling dtoverlay parameter **dtoverlay = waveshare35b** and **dtoverlay = waveshare35b: rotate = 90, swapxy = 1**
Save and Reboot.

The solution completely fixed the problem (color inversion on LCD35B rev.1) in an bonus, now you do not need to manually correct the dual screen and screen size!

2. issues: UPS Backup-UPS 700 I get a message on the screen: ups service but not running

An attempt to turn on the [FAQ](https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#how-to-connect-a-ups-to-the-raspiblitz) manual, unsuccessful
In addition, I had to do
`sudo apt --fix-broken install`
After which the power-on command worked fine
`/home/admin/config.scripts/blitz.ups.sh on apcusb`

rootzoll, 2021-04-29:
Yeah we needed to change LCD drivers with the 64-bit OS ... and at the moment we dont have the capacity to support other LCDs beside the one in the hardware list. Thanks for sharing the fix for your LCD hat, but we might not add this as a default to the code for now and future updates with new sd card might brak your changes again. Maybe you can write your fixes into the `/mnt/hdd/app-data/custom-installs.sh` file that will be excuted afetr every recovery/update.

rootzoll, 2021-04-29:
For the v1.7.1 I will mark this issue to check that the UPS script gets activated on recovery/install if set.

intorid, 2021-04-30:
> Maybe you can write your fixes into the `/mnt/hdd/app-data/custom-installs.sh` file that will be excuted afetr every recovery/update.

Thanks for the comprehensive answers, your script for the next updates via **custom-installs.sh** will be fine. 
The question can be closed!

closed 2021-04-30 by intorid