raspiblitz/raspiblitz #2247: Not able to get HTTPS response

thecharacter, 2021-04-28:
I upgraded to v1.7 and got this error. This is specially happening with the LNbits webinterface order. I reinstalled LNbits but getting the same error. I'm able to access LNBits with the https://xxxxx.onion address and with my LetsEncrypt domain but services such as LNurlp and LNurlw the require HTTPS communication beyond my local network fail. 

Note: I updated Tor per the repair menu and reinstalled LNbits. 

[image]

rootzoll, 2021-04-28:
Thankls for the report. Have you already closed all subscriptions (IP2Tor & LetsEncrypt) and renewed them also. 

Note: This might give you a new port and break links that you already distributed in your local community.

@frennkie can you also take a look here?

thecharacter, 2021-04-28:
@rootzoll yeah I closed everything and created new subscriptions. Still getting this error.

rootzoll, 2021-04-29:
Hmm .. I added I bit more debug info to the v1.7 code. Please go to MAINMENU > UPDATE > PATCH > PATCH-MENU > PATCH (you dont need to reboot) and then please take a picture again of the Warning window.

thecharacter, 2021-04-29:
Cool here it is. 

[image]

rootzoll, 2021-04-30:
Can you EXIT to terminal and run the command `/home/admin/config.scripts/internet.dyndomain.sh off` and reboot and let me know if that changed someting.

thecharacter, 2021-04-30:
Ran the command rebooted and got the same warning. I deleted the subscriptions and created new subscriptions and got same warning. 

[image]

klnit, 2021-05-01:
I have the same issue and think it is the reason for #2240
Please link it to this issue

frennkie, 2021-05-01:
This error message seems to come from here: https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/blitz.subscriptions.letsencrypt.py#L751 which is called from here

```
config.scripts/bonus.lnbits.sh:123:        domainWarning=$(sudo /home/admin/config.scripts/blitz.subscriptions.letsencrypt.py subscription-detail ${domain} ${port} | jq -r ".warning")
``` 

@rootzoll is this more than a warning? As far as I see it the actual "IP2Tor" services is working fine here..

JurGi7, 2021-05-04:
Hello, I have the same or similar problem, since the update to 1.7.0 my LetsEncrypt IP2TOR subscription no longer works. This message comes up briefly and I don't have an entry under subscription 🤔. 
[
[image]
](url)

rootzoll, 2021-05-13:
hey @thecharacter I merged a possible fix to the v1.7 branch. Can you please test if this fixes this for you with first updating to that oatch with MAINMENU > UPDATE > PATCH > PATCHMENU > PATCH and then do a reboot.

Then under SUBSCRIVE > LIST remove the LetsEncrypt and then re-add it fresh.

JurGi7, 2021-05-18:
in my case - patch installed, everything works again. thanks rootzoll

thecharacter, 2021-05-19:
It's working thank you.

peterbaricic, 2021-07-07:
Hi,
I have the same issue (not able to get https response), but also when I try to unsubscribe from ip2tor and letsencrypt, ip2tor unsubscribe is working, but unsubscribing letsencrypt gives me an error='letsencrypt acme failed'

maddadder, 2021-07-13:
> Hi,
> I have the same issue (not able to get https response), but also when I try to unsubscribe from ip2tor and letsencrypt, ip2tor unsubscribe is working, but unsubscribing letsencrypt gives me an error='letsencrypt acme failed'

I also got the above error and was able to fix: 'letsencrypt acme failed' by editing /mnt/hdd/raspiblitz.conf and changing from letsencrypt=on to letsencrypt=off and then attempted to unsubscribe and it worked

peterbaricic, 2021-07-13:
> I also got the above error and was able to fix: 'letsencrypt acme failed' by editing /mnt/hdd/raspiblitz.conf and changing from letsencrypt=on to letsencrypt=off and then attempted to unsubscribe [...]

I have letsencrypt=off and it still gives me the same error

maddadder, 2021-07-13:
> I have letsencrypt=off and it still gives me the same error

Wait. I think I got it mixed up. Try setting it to on. That way it can turn it off. Sorry about the confusion.

peterbaricic, 2021-07-13:
> Wait. I think I got it mixed up. Try setting it to on. That way it can turn it off. Sorry about the confusion.

that worked, thanks

rootzoll, 2021-12-07:
This error message `Target is not HTTPS` comes when you try to make an IP2Tor subscription, but the onion service is not responding.

For future versions have a script that checks reachability of an onion service BEFORE trying to create an IP2Tor subscription .. for example with: torsocks curl http://soveng42aj6ynynkqeyut2dwnmwou45epfpuwvjy2t45fpxn3dk2ymad.onion

If a non working onion-service is detected the user should be offered to renew the onion hidden service BEFORE trying to make the IP2Tor bridge. Because its some re-work .. will push this to v1.8.0

kevkevinpal, 2021-12-08:
[image]

is this error related to this issue? It is rejecting my subscription when trying to subscribe with IP2Tor

rootzoll, 2021-12-08:
> Does this mean this happens when the tor service for what we want to bridge is not working or the ip2tor shop service isn't working?

@kevkevinpal in the case I had on my test machine it was because the tor onion service on my blitz was not reachable. The IP2Tor service makes a test call to the service before creating an invoice to check if it gets an HTTPS response but this fails - it gets no response at all and it gives back this error. You can test this by going into MAINMENU > SPHINX and there you copy the onion address and check if you can call with a normal Tor browser (with prefix `http://` or `https://`). Can you test on your device?

rootzoll, 2022-06-02:
I also had this error again on one of my test machines .. even tho my tor service with https was working when I call it from my desktop tor browser.

Mexxblitz, 2022-11-05:
[image]

Is this also Related to the topic? Constantly getting this error message. 
What is happening in the background and is there a way to fix it?

Here the Error Syntax: 
###### ERROR DETAIL FOR DEBUG #######

Error Short:
timeout on getting invoice
Shop:
fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion
Bridge:
{'id': 'c62cc284-7f56-4f77-a583-cec3fc905dc7', 'site': 'ip2tor.fulmo.org', 'created_at': '2022-01-14T17:50:59Z', 'modified_at': '2022-11-05T12:10:19Z', 'ip': '46.165.236.103', 'is_enabled': True, 'is_alive': True, 'name': 'eu-fra3', 'is_testnet': False, 'offers_tor_bridges': True, 'tor_bridge_duration': 86400, 'tor_bridge_price_initial': 28000, 'tor_bridge_price_extension': 18000, 'offers_rssh_tunnels': False, 'rssh_tunnel_price': 1000, 'terms_of_service': '-', 'terms_of_service_url': '-', 'ci_date': '2022-11-05T12:10:19Z', 'ci_message': '', 'ci_status': 0, 'owner': 6, 'tor_bridge_duration_hours': 24, 'tor_bridge_price_initial_sats': 28, 'tor_bridge_price_extension_sats': 18}
Error Detail:
{'url': 'http://fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion/api/v1/public/pos/1df168a8-a94e-49d3-b33b-433374a44a5e/', 'status': 'M', 'message': None, 'item_details': [{'url': 'http://fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion/api/v1/public/po_items/7cb782f4-fde2-4250-961b-9a6c09b5056c/', 'product_id': 'a2079bab-7ef2-4dfc-b041-fb6bf02bdce6', 'product': {'url': 'http://fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion/api/v1/tor_bridges/a2079bab-7ef2-4dfc-b041-fb6bf02bdce6/', 'id': 'a2079bab-7ef2-4dfc-b041-fb6bf02bdce6', 'comment': 'RaspiBlitz', 'status': 'I', 'host': {'url': 'http://fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion/api/v1/hosts/c62cc284-7f56-4f77-a583-cec3fc905dc7/', 'ip': '**(...)**', 'name': 'eu-fra3', 'site': {'domain': 'ip2tor.fulmo.org', 'name': 'ip2tor.fulmo.org'}, 'is_testnet': False, 'ci_date': '2022-11-05T12:13:35Z', 'ci_status': 0, 'ci_message': ''}, 'port': 45762, 'target': 'kno5a4duvvo2nswsn6oex5ibqj3cyk53dpuq7wwfqraobpcamo6nbvid.onion:443', 'suspend_after': '2022-11-06T12:20:39Z'}, 'position': 0, 'price': 28000, 'quantity': 1, 'po': 'http://fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion/api/v1/public/pos/1df168a8-a94e-49d3-b33b-433374a44a5e/'}], 'ln_invoices': [], 'timestamp': '2022-11-05T12:13:46Z'}

Press Enter to continue ...

cstenglein, 2023-06-07:
IP2Tor is dead for now => closed

closed 2023-06-07 by cstenglein