raspiblitz/raspiblitz #2249: MInimize / confirm restarts after installing services

openoms, 2021-04-29:
When running a routing node any downtime should be minimized.
Restarting the whole node should be the last resort.

I often only use the install scripts in cli to avoid restarts from the menu.
Some services now start automatically in the `raspiblitz.info` has the `state=ready`.

Should aim to modify the scripts to not need full restarts and at least have a chance to confirm before forcing it.
"Some services will only work after restart."

nyxnor, 2021-06-22:
Tried to change passwords A, B, C. They all asked to restart, but for sure C just needs to restart LND.
Also, changing password C does not let you exit the menu screen, you are stuck there, `exit` button does not work.

rootzoll, 2021-07-01:
The new setup process also tries to do the setup in one go without restart. Because a WebUI does not wrok well with restarts I put this on the list for v1.8 that password resetting should not require restarts anymore.

rootzoll, 2022-03-09:
v1.7.2 needed one setup-restart in the end .. so I we need also have the WebUI being able to handle restarts.

openoms, 2022-05-02:
Still restarting after activating the Testnets. It can be avoided.

rootzoll, 2022-05-09:
I think the amount of restart in general was reduced quite good since this issue was created ... closing now. If there is still a restart that somebody likes to see go away please open a new ssue on every single one.

closed 2022-05-09 by rootzoll