raspiblitz/raspiblitz #2250: Fast update option - inactivate services before switching off for an SDcard update

openoms, 2021-04-29:
The downtime of routing nodes should be minimized.

When changing the SDcard with many services active (especially the lengthy compilation of Electrs, RTL, JM) the node is down unnecessarily.

There could be an option in the UPDATE menu to deactivate services before switching off for the update (edit most services to `off` in the `raspiblitz.conf`).
This would prevent the the long compilation time before starting the lightning node and services can be installed again while running. (Preferably without further restarts as discussed in: https://github.com/rootzoll/raspiblitz/issues/2249)

The simplest solution is to just leave it to the advanced user to install the services manually after using the "fast update" option or could delay the provision script also (adding more complexity).

rootzoll, 2021-04-29:
I think thats easier done by keeping that in mind with the re-factor of the non-interactive setup/update process - see #1126 I wil make a note there - that if possible the the bitcoin & lnd should start before the additional apps get installed.

I think that will slow down the recover/update process in general ... but I see the importance of keeping lightning down-time as small possible.

rootzoll, 2021-04-29:
Ah OK .. you already did that ;)

rootzoll, 2021-04-29:
Because its now mentioned in the #1126  .. can we close this issue?

openoms, 2021-04-29:
Sure

closed 2021-04-29 by openoms