raspiblitz/raspiblitz #2252: acpusb and lnbits not installing properly while updating to 1.7.0

m00ninite, 2021-04-29:
Here's the logs coming from raspiblitz.log while running the update install scripts:

[log.log](https://github.com/rootzoll/raspiblitz/files/6400381/log.log)

nyxnor, 2021-04-29:
The reason why it didnt get the apcupsd
```
The following NEW packages will be installed:
  apcupsd apcupsd-doc s-nail
0 upgraded, 3 newly installed, 0 to remove and 14 not upgraded.
Need to get 1,355 kB of archives.
After this operation, 2,833 kB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.
Failed to stop apcupsd.service: Unit apcupsd.service not loaded.
```
This ok shouldnt occur:
```
Failed to start apcupsd.service: Unit apcupsd.service not found.
OK - UPS is now connected
```

The lnbits error:
```
Building wheels for collected packages: brotli, markupsafe, pyscss, sqlalchemy
  Running setup.py bdist_wheel for brotli: started
  Running setup.py bdist_wheel for brotli: finished with status 'error'
  Complete output from command /home/lnbits/lnbits/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-omcdw7wk/brotli/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-_c8wrfud --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
```

rootzoll, 2021-04-30:
@nyxnor good catch on the apcupsd 

@m00ninite is LNbits just not installing during update-process or does it also not work withthe same error when you de- & install it manually from the SERVICES menu?

m00ninite, 2021-05-03:
@rootzoll So LNbits _did_ install, I just think it wasn't able to fully install all the plugins, so it may be somewhat broken. A manual uninstall/reinstall also came up with the same error.

rootzoll, 2021-09-20:
To be tsted with v1.7.1rc2.

frennkie, 2021-10-12:
I was just updating vor 1.7.1 and also got several errors regarding bdist_wheel during the LNbits installation.

```
Building wheels for collected packages: brotli, markupsafe, pyscss, sqlalchemy
  Running setup.py bdist_wheel for brotli: started
  Running setup.py bdist_wheel for brotli: finished with status 'error'
  Complete output from command /home/lnbits/lnbits/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-in                                                       stall-hwae44ti/brotli/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();ex                                                       ec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-8stgcod3 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for brotli
  Running setup.py clean for brotli
  Running setup.py bdist_wheel for markupsafe: started
  Running setup.py bdist_wheel for markupsafe: finished with status 'error'
  Complete output from command /home/lnbits/lnbits/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-in                                                       stall-hwae44ti/markupsafe/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close(                                                       );exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-himk362b --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for markupsafe
  Running setup.py clean for markupsafe
  Running setup.py bdist_wheel for pyscss: started
  Running setup.py bdist_wheel for pyscss: finished with status 'error'
  Complete output from command /home/lnbits/lnbits/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-in                                                       stall-hwae44ti/pyscss/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();ex                                                       ec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-gz90xzef --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for pyscss
  Running setup.py clean for pyscss
  Running setup.py bdist_wheel for sqlalchemy: started
  Running setup.py bdist_wheel for sqlalchemy: finished with status 'error'
  Complete output from command /home/lnbits/lnbits/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-in                                                       stall-hwae44ti/sqlalchemy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close(                                                       );exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-4xl55nfv --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for sqlalchemy
  Running setup.py clean for sqlalchemy
Failed to build brotli markupsafe pyscss sqlalchemy
Installing collected packages: aiofiles, async-generator, attrs, bech32, bitstring, blinker, brotli, cerberus, certifi, clic                                                       k, six, ecdsa, embit, marshmallow, python-dotenv, environs, h11, hyperframe, hpack, h2, sniffio, httpcore, rfc3986, httpx, w                                                       sproto, toml, priority, typing-extensions, hypercorn, idna, itsdangerous, markupsafe, jinja2, pydantic, lnurl, outcome, pypn                                                       g, pyqrcode, pyscss, werkzeug, quart, quart-compress, quart-cors, sortedcontainers, trio, quart-trio, represent, secure, sho                                                       rtuuid, sqlalchemy, sqlalchemy-aio
  Running setup.py install for brotli: started
```

I guess the LNbits venv needs an additional installation step.

krtk6160, 2021-11-09:
I am also seeing the same `bdist_wheel` error as other users above while trying to install lnbits on raspi 1.7.1

openoms, 2021-11-09:
Would it help if we would use a python venv for LNbits?

rootzoll, 2021-11-30:
Cannot reproduce on my test machines. Waiting for more reports in this. But looking into venv for LNbits is an option to look into for a later relase like 1.8.0

rootzoll, 2022-05-09:
I am closing this issue because there were no further reports

closed 2022-05-09 by rootzoll