raspiblitz/raspiblitz #2256: Litecoin Install Froze at Synchronize Blockchain

ronledfordjr, 2021-04-30:
[image]

I let it run overnight and no further movement since last night.

BTW, my node for Bitcoin is rocking along and seems, to me at least, a lot faster than the set up of 1.63 - thanks.

I am updating node to Version 1.7.0 with lnd 0.12.1 and bitcoin 0.21.0 (or litecoin 0.18.1) from 1.63 - fresh instal

For this node I chose Litecoin

Raspberry Pi 4B 8BG
SSD 1TB
SD 32GB (format FAT) I had it on hand

https://termbin.com/lpr9

debug log

[H[2J[3J
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.0
chainnetwork: litecoin / main
 10:35:14 up  9:29,  2 users,  load average: 0.29, 0.34, 0.40

*** BLOCKCHAIN SYSTEMD STATUS ***
● litecoind.service - Litecoin daemon
   Loaded: loaded (/etc/systemd/system/litecoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-04-30 00:55:23 BST; 9h ago
  Process: 1387 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1393 ExecStart=/usr/local/bin/litecoind -daemon -conf=/home/bitcoin/.litecoin/litecoin.conf -pid=/home/bitcoin/.litecoin/litecoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1413 (litecoind)
    Tasks: 17 (limit: 4915)
   CGroup: /system.slice/litecoind.service
           └─1413 /usr/local/bin/litecoind -daemon -conf=/home/bitcoin/.litecoin/litecoin.conf -pid=/home/bitcoin/.litecoin/litecoind.pid

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u litecoind -b --no-pager -n8
-- Logs begin at Fri 2021-04-30 02:59:13 BST, end at Fri 2021-04-30 10:35:15 BST. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/litecoin/debug.log
2021-04-30T08:36:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 30600 seconds ago)
2021-04-30T08:39:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 30780 seconds ago)
2021-04-30T08:42:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 30960 seconds ago)
2021-04-30T08:45:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 31140 seconds ago)
2021-04-30T08:48:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 31320 seconds ago)
2021-04-30T08:51:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 31500 seconds ago)
2021-04-30T08:54:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 31680 seconds ago)
2021-04-30T08:57:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 31860 seconds ago)
2021-04-30T09:00:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32040 seconds ago)
2021-04-30T09:03:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32220 seconds ago)
2021-04-30T09:06:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32400 seconds ago)
2021-04-30T09:09:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32580 seconds ago)
2021-04-30T09:12:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32760 seconds ago)
2021-04-30T09:15:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32940 seconds ago)
2021-04-30T09:18:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 33120 seconds ago)
2021-04-30T09:21:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 33300 seconds ago)
2021-04-30T09:24:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 33480 seconds ago)
2021-04-30T09:27:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 33660 seconds ago)
2021-04-30T09:30:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 33840 seconds ago)
2021-04-30T09:33:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 34020 seconds ago)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
[... 196 lines cut ...]
*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.115'
tempCelsius='47.7'
uptime=34155
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=0
initialSync=1
syncProgress=0.00 
startcountLightning=1
lndActive=1
walletLocked=0
lndPeers=0
syncedToChain=0
scanTimestamp=1317972665
scanDate='Fri  7 Oct 08:31:05 BST 2011'
scanProgress=22.38
lndRPCReady=1
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

ronledfordjr, 2021-04-30:
I went ahead and started over again. It is advancing now.

I logged in as pi, I have never done this before, and I could not find the means to kill this session. Was it causing the issue mentioned above?
[image]

nyxnor, 2021-04-30:
This was the problem, it was mounted wrong:
```
Filesystem Type Size Used Avail Use% Mounted on
/dev/sdb2 btrfs 868G 21M 866G 1% /mnt/storage
/dev/sdb1 btrfs 30G 28M 28G 1% /mnt/hdd
```
Login as admin, pi is more watch-only.

ronledfordjr, 2021-05-01:
Thank youi

closed 2021-05-01 by ronledfordjr