raspiblitz/raspiblitz #2260: Stuck, Litecoin Node LND waiting for synch - attached LND Log

ronledfordjr, 2021-05-01:
[image]

I was adding services to this node this morning.
Run behind TOR
Circuitbreaker
LND Auto-Unlock
StationChannelBackup on DropBox
UPnP for both BTC and LND

I no longer see UPnP to deselect

Other the Run behind TOR I have deselected.

RPi 4B 8GB
1TB WD SSD

Will follow the /mnt/hdd/litecoin/debug.log
running: 'sudo tail -n 30 -f /mnt/hdd/litecoin/debug.log'

Press ENTER to continue
use CTRL+C any time to abort .. then use command 'raspiblitz' to return to menu
###############################################################################

2021-05-01T13:59:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 7020 seconds ago)
2021-05-01T14:02:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 7200 seconds ago)
2021-05-01T14:05:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 7380 seconds ago)
2021-05-01T14:08:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 7560 seconds ago)
2021-05-01T14:11:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 7740 seconds ago)
2021-05-01T14:14:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 7920 seconds ago)
2021-05-01T14:17:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 8100 seconds ago)
2021-05-01T14:20:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 8280 seconds ago)
2021-05-01T14:23:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 8460 seconds ago)
2021-05-01T14:26:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 8640 seconds ago)
2021-05-01T14:29:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 8820 seconds ago)
2021-05-01T14:32:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 9000 seconds ago)
2021-05-01T14:35:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 9180 seconds ago)
2021-05-01T14:38:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 9360 seconds ago)
2021-05-01T14:41:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 9540 seconds ago)
2021-05-01T14:44:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 9720 seconds ago)
2021-05-01T14:47:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 9900 seconds ago)
2021-05-01T14:50:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 10080 seconds ago)
2021-05-01T14:53:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 10260 seconds ago)
2021-05-01T14:56:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 10440 seconds ago)
2021-05-01T14:59:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 10620 seconds ago)
2021-05-01T15:02:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 10800 seconds ago)
2021-05-01T15:05:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 10980 seconds ago)
2021-05-01T15:08:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 11160 seconds ago)
2021-05-01T15:11:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 11340 seconds ago)
2021-05-01T15:14:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 11520 seconds ago)
2021-05-01T15:17:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 11700 seconds ago)
2021-05-01T15:20:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 11880 seconds ago)
2021-05-01T15:23:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 12060 seconds ago)
2021-05-01T15:26:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 12240 seconds ago)

ronledfordjr, 2021-05-01:
This is my BTC LND Node with same adds and working well

[image]

nyxnor, 2021-05-01:
Can you try connecting manually to more peers? I dont know the commands or where to find them when using lite.

ronledfordjr, 2021-05-01:
I ran the debug log

https://termbin.com/bx8k

[H[2J[3J
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.0
chainnetwork: litecoin / main
 21:41:02 up  4:42,  2 users,  load average: 0.47, 0.53, 0.51

*** BLOCKCHAIN SYSTEMD STATUS ***
● litecoind.service - Litecoin daemon
   Loaded: loaded (/etc/systemd/system/litecoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-05-01 16:58:37 BST; 4h 42min ago
  Process: 1604 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1610 ExecStart=/usr/local/bin/litecoind -daemon -conf=/home/bitcoin/.litecoin/litecoin.conf -pid=/home/bitcoin/.litecoin/litecoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1642 (litecoind)
    Tasks: 18 (limit: 4915)
   CGroup: /system.slice/litecoind.service
           └─1642 /usr/local/bin/litecoind -daemon -conf=/home/bitcoin/.litecoin/litecoin.conf -pid=/home/bitcoin/.litecoin/litecoind.pid

May 01 16:58:37 raspberrypi systemd[1]: Starting Litecoin daemon...
May 01 16:58:37 raspberrypi systemd[1]: Started Litecoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u litecoind -b --no-pager -n8
-- Logs begin at Sat 2021-05-01 16:57:59 BST, end at Sat 2021-05-01 21:41:02 BST. --
May 01 16:58:37 raspberrypi systemd[1]: Starting Litecoin daemon...
May 01 16:58:37 raspberrypi systemd[1]: Started Litecoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/litecoin/debug.log
2021-05-01T19:41:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 13320 seconds ago)
2021-05-01T19:44:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 13500 seconds ago)
2021-05-01T19:47:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 13680 seconds ago)
2021-05-01T19:50:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 13860 seconds ago)
2021-05-01T19:53:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 14040 seconds ago)
2021-05-01T19:56:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 14220 seconds ago)
2021-05-01T19:59:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 14400 seconds ago)
2021-05-01T20:02:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 14580 seconds ago)
2021-05-01T20:05:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 14760 seconds ago)
2021-05-01T20:08:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 14940 seconds ago)
2021-05-01T20:11:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 15120 seconds ago)
2021-05-01T20:14:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 15300 seconds ago)
2021-05-01T20:17:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 15480 seconds ago)
2021-05-01T20:20:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 15660 seconds ago)
2021-05-01T20:23:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 15840 seconds ago)
2021-05-01T20:26:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 16020 seconds ago)
2021-05-01T20:29:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 16200 seconds ago)
2021-05-01T20:32:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 16380 seconds ago)
2021-05-01T20:35:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 16560 seconds ago)
2021-05-01T20:38:36Z Potential stale tip detected, will try using extra outbound peer (last tip update: 16740 seconds ago)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-05-01 16:58:37 BST; 4h 42min ago
  Process: 1643 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 1645 (lnd)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/lnd.service
           └─1645 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --tor.socks=9070 --tor.control=9071 --listen=127.0.0.1:9735

May 01 16:58:37 raspberrypi systemd[1]: Starting LND Lightning Daemon...
May 01 16:58:37 raspberrypi systemd[1]: Started LND Lightning Daemon.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
[... 193 lines cut ...]
tempCelsius='51.1'
uptime=16970
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=2044222
initialSync=0
syncProgress=99.96 
startcountLightning=1
lndActive=1
walletLocked=0
lndPeers=0
syncedToChain=0
scanTimestamp=1619832071
scanDate='Sat  1 May 02:21:11 BST 2021'
scanProgress=99.98
lndRPCReady=1
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

ronledfordjr, 2021-05-01:
I reset the LND Wallet since I had no coins in it yet.

nyxnor, 2021-05-01:
Could you please post only the link or the logs? As the logs are inside the link.
Have you tried adding peers?

ronledfordjr, 2021-05-01:
I could be picking all the closed nodes. Here are the results with a screengrab of one of the nodes from 1ml. Do I need to fund wallet first before connecting?

[image]

[image]

[image]

[image]

nyxnor, 2021-05-01:
Not a channel, a peer, for litecoin, not for lightning.

ronledfordjr, 2021-05-03:
I reset LND Wallet

closed 2021-05-03 by ronledfordjr