raspiblitz/raspiblitz #2261: Multi user setup

chkpnt, 2021-05-02:
Sorry, can't find anything to this in the readme and faq: Can I use one Raspiblitz for multiple wallets? Or is a multi-user setup more Apple like and I would need one Raspiblitz for each family member?

rootzoll, 2021-05-02:
The RaspiBlitz itself is just a for one Admin ... but check software like LNbits that is like a sub-account management system for your Lightning node and of course you can offer basic services to your family like providing a Electrum server or have them running a lightning wallet on the computer or mobile phone that just needs one channel to your RaspiBlitz and you provide good connectivity with multiple channels and an extra layer of privacy to them.

chkpnt, 2021-05-02:
Okay, thanks. I don't own a RaspiBlitz yet., I'm just trying to figure out if it would fit my thoughts. But as I'm not very experienced in this space, the thoughts themselves may be wrong.

For Lightning, I'm currently using a custodial BlueWallet, which can be easily be funded by an on-chain transaction. For the backend, a self-hosted LNDHub instance can be used, and I just read that LNbits has a LNDHub integration: So maybe this is exactly what I'm looking for: the family members could connect to the LNDHub instance on RaspiBlitz and create Lightning wallets and fund them by their own. And I'm only responsible for the initial channel setup and the backups.

I'll read more about the software bundled with RaspiBlitz, thanks for the hints.

closed 2021-05-02 by chkpnt